MichaelGrupp/evo #1: How to get different sequences of KITTI datasets(KITTI01, .. , 10)

haidela, 2017-11-01:
Hello, Based on this explanation [https://github.com/raulmur/ORB_SLAM2/issues/454](url) I am trying to do sequences KITTI01,02,.....,10 for S-PTAM, ORB_SLAM and LSD-SLAM 2 days ago unsuccessfully
@MichaelGrupp Can you help me to explain the procedure?
Thank you in advance for your answer.

MichaelGrupp, 2017-11-01:
Please ask more specific what doesn't work for you, what you tried already etc.

haidela, 2017-11-01:
@MichaelGrupp I am trying to do KITTI_01_SPTAM.txt, KITTI_02_ORB.txt, KITTI_01_LSD_SLAM.txt(other different sequences (01, 02, ..., 10) of the sequence 00 of KITTI dataset), I don't understand your explanation:Since you want to use TUM RGB-D, KITTI and EuRoC datasets keep in mind that the "default" ground truth trajectory file formats of these datasets are different:
TUM: timestamp + xyz + quaternion per row
KITTI: flattened SE(3) pose matrix per row (without bottom row 0 0 0 1)
EuRoC: csv with state information in IMU/body frame
Please, can you explain a little the procedure ?

MichaelGrupp, 2017-11-01:
> KITTI_01_SPTAM.txt, KITTI_02_ORB.txt, KITTI_01_LSD_SLAM.txt

What are those files? Which format? And *what* exactly do you want to achieve?

My explanation was just a reminder that there are different trajectory file formats for the ground truth / reference trajectory depending on the dataset.

haidela, 2017-11-02:
@MichaelGrupp Hello, I want to make trajectories( in text file format like yours  KITTI_00_Orb.text, KITTI_00_SPTAM) performed by S-PTAM and Orb_Slam on KITTI diferent sequences of datasets with my camera parameters. 
My test steps for ORB_SLAM(based on this issue [https://github.com/raulmur/ORB_SLAM/issues/28](url):
1. roscore
2.  roslaunch ExampleGroovyOrNewer.launch
3. rosbag play --pause data/sequence01.bag
4. python lsd_to_readable.py time.txt calib.txt (time.txt and calib.txt performed by the sequence of KITTI dataset)

SUMMARY
========
PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
NODES
auto-starting new master
process[master]: started with pid [18477]
ROS_MASTER_URI=http://haidara-virtual-machine:11311/

$ rostopic list
/ORB_SLAM/Frame
/ORB_SLAM/Map
/ORB_SLAM/Map_array
/camera/image_raw
/clicked_point
/clock
/image_view/output
/image_view/parameter_descriptions
/image_view/parameter_updates
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static

I didn't successfully tested.
Will you explain  your procedure to me?

haidela, 2017-11-02:
To get the trajectories in text file format

MichaelGrupp, 2017-11-02:
1. Your question is about using the implementations of SLAM algorithms that are not maintained by me and aren't part of this repository. If you need help with running one of those, please open an issue in the repository of the SLAM algorithm you want to use. This will also increase the chances that other users see it who might be able to help you.

Please understand that I only answer specific issues with *my* software here. 

2. One tip: looking at your infos, it looks like you're using the "old" monocular-only implementation of ORB-SLAM ([this one](https://github.com/raulmur/ORB_SLAM/commits/master)). ORB-SLAM2 is the updated version for monocular, RGB-D and stereo cameras - see [here](https://github.com/raulmur/ORB_SLAM2). The README also explains how to use the KITTI dataset. The pose file you will get from this example can be compared to the ground truth (download [here](http://www.cvlibs.net/datasets/kitti/eval_odometry.php)) using my evo package.

haidela, 2017-11-02:
Thank you very much for your valued assistance and your quick reply, which helped me to understand many things

MichaelGrupp, 2017-11-02:
You're welcome.

closed 2017-11-02 by MichaelGrupp