MichaelGrupp/evo #3: how to compare diferent kind trajectories?

cumtchenchang, 2017-11-07:
I am so sorry to ask this question, maybe it is  samplie. I want to konw how to compare diferent kind trajectories? such as euroc trajectory and kitti trajectory. or how to translate one kind to another kind? for example, translate  euroc trajectory and kitti trajectory.
                                                                    thank you.

MichaelGrupp, 2017-11-07:
Can you upload one example trajectory file for each of the formats here? This makes it easier for me to help you.

Just drag the files into the comment box.

MichaelGrupp, 2017-11-07:
In general, you can export trajectories to other formats with `evo_traj`using the `--save_as_<other_format>` option.

The following combinations are possible:

|       | `--save_as_bag`    | `--save_as_kitti` | `--save_as_tum`    |
|-------|:--------------------:|:-------------------:|:--------------------:|
| `bag`   | yes                | yes               | yes                |
| `euroc` | yes                | yes               | yes                |
| `kitti` | no (no timestamps) | yes               | no (no timestamps) |
| `tum`   | yes                | yes               | yes                |

Example:
```bash
# export a EuRoC groundtruth file to a TUM trajectory
evo_traj euroc data.csv --save_as_tum
# (will be saved as data.tum)

# export TUM trajectories to KITTI format
evo_traj tum traj_1.txt traj_2.txt traj_3.txt --save_as_kitti
# (will be saved as *.kitti)

# export TUM trajectories to ROS bagfile
evo_traj tum traj_1.txt traj_2.txt traj_3.txt --save_as_bag
# (will be saved as <timestamp>.bag with topics traj_1, traj_2 and traj_3)

# and so on...
```

There is no `--save_as_euroc`option because the EuRoC format only makes sense for the ground truth of the EuRoC dataset.

MichaelGrupp, 2017-11-07:
It's actually good that you ask, because I don't use EuRoC files very often and found a bug now :smile: 

There was a typo in the file reader, I fixed it in this commit: https://github.com/MichaelGrupp/evo/commit/653e735f8c78852d71d1c86a4d9b74b4d561ea50

Now the `euroc` mode should work fine, let me know if there are problems. You need to run `pip install evo --upgrade` if you installed from PyPi or `git pull` if you installed from source.

cumtchenchang, 2017-11-08:
thank for your help, I am so greatful for you. I am srearching SLAM especially visual inertial odometry. your code plays an important role in my study.

cumtchenchang, 2017-11-08:
I do not know why the Y-Axis of evo_res and  evo_ape result plot  are also APE. maybe one is wrong.

cumtchenchang, 2017-11-08:
I am so sorry to disturb you. Can evo_rpe or evo_rpe-for-each just need est_file? many dataset like my moblie robot's dataset do not have ground truth.I want to compare three trajetory' RPE from one dataset, but do not konw how to use your code.
                                            thank you 
[my_trajectory.zip](https://github.com/MichaelGrupp/evo/files/1453221/my_trajectory.zip)

MichaelGrupp, 2017-11-08:
Theoretically, you can replace `ref_file` by any trajectory you want. If it makes sense for you, this trajectory doesn't have to be a real ground truth. You can do that to compare trajectories if you want, however these metrics are designed to be used with ground truth normally.

By the way: if you use Linux you can clone the repository and go to `evo/evo/test/demos` - there are some "tutorial" shell scripts you can execute that can help you to understand the software. I am trying to improve the documentation when I have time, but for now you can use that.

cumtchenchang, 2017-11-08:
thank you very much. I follow you said now

cumtchenchang, 2017-11-08:
see your typo, three hours ago, I reinstall the cold. Then, I can not use any function with --plot.
I also tried other functions such as evo_rpe、evo_apt, this problem also appeared.
cc@cc-B250-HD3:~/code/evo/evo/test/data$ evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
--------------------------------------------------------------------------------
loaded 4541 poses from: KITTI_00_ORB.txt
loaded 4541 poses from: KITTI_00_SPTAM.txt
loaded 4541 poses from: KITTI_00_gt.txt
--------------------------------------------------------------------------------
name:	KITTI_00_ORB
infos
	nr. of poses	4541
	path length (m)	3705.09773346
	pos_end (m)	[ -6.25027037  -0.92649209  94.90350342]
	pos_start (m)	[ -4.00000000e-09   0.00000000e+00   0.00000000e+00]
--------------------------------------------------------------------------------
name:	KITTI_00_SPTAM
infos
	nr. of poses	4541
	path length (m)	3718.50094293
	pos_end (m)	[-10.90358769  -1.23925455  99.43272323]
	pos_start (m)	[ 0.  0.  0.]
--------------------------------------------------------------------------------
name:	KITTI_00_gt
infos
	nr. of poses	4541
	path length (m)	3724.1869906
	pos_end (m)	[ -5.583931  -3.562758  96.96153 ]
	pos_start (m)	[  5.55111500e-17   3.33066900e-16  -4.44089200e-16]
[ERROR]
unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 100, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_traj.py", line 287, in run
    plt.tight_layout()
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 1323, in tight_layout
    fig.tight_layout(pad=pad, h_pad=h_pad, w_pad=w_pad, rect=rect)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/figure.py", line 2007, in tight_layout
    pad=pad, h_pad=h_pad, w_pad=w_pad, rect=rect)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/tight_layout.py", line 351, in get_tight_layout_figure
    pad=pad, h_pad=h_pad, w_pad=w_pad)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/tight_layout.py", line 130, in auto_adjust_subplotpars
    if ax.get_visible()])
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axes/_base.py", line 3960, in get_tightbbox
    bb_xaxis = self.xaxis.get_tightbbox(renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axis.py", line 1092, in get_tightbbox
    renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axis.py", line 1075, in _get_tick_bboxes
    extent = tick.label1.get_window_extent(renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/text.py", line 970, in get_window_extent
    bbox, info, descent = self._get_layout(self._renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/text.py", line 354, in _get_layout
    ismath=False)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_agg.py", line 226, in get_text_width_height_descent
    s, fontsize, renderer=self)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/texmanager.py", line 602, in get_text_width_height_descent
    dvifile = self.make_dvi(tex, fontsize)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/texmanager.py", line 393, in make_dvi
    stderr=subprocess.STDOUT)
  File "/usr/local/lib/python2.7/dist-packages/subprocess32.py", line 629, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/subprocess32.py", line 825, in __init__
    restore_signals, start_new_session)
  File "/usr/local/lib/python2.7/dist-packages/subprocess32.py", line 1574, in _execute_child
    raise child_exception_type(errno_num, err_msg)
OSError: [Errno 2] No such file or directory: 'latex'

[ERROR]
evo module evo.main_traj crashed - no logfile written (disabled)

MichaelGrupp, 2017-11-08:
Did you change the `plot_usetex`setting to true? If yes, you would need LaTeX installed.

To get rid of the error, try `evo_config set plot_usetex false` or `evo_config reset` (may require `sudo`).

cumtchenchang, 2017-11-08:
I solved the problem with  sudo evo_config reset. maybe I miused plot_usetex

MichaelGrupp, 2017-11-08:
Could be that you used the latex_demo.sh, I think I added a config change there - probably not your mistake.

closed 2017-11-11 by MichaelGrupp