MichaelGrupp/evo #5: Running error

haidela, 2017-11-13:
Hello everyone, I want to get ape performed by LSD_Slam on EuRoC diferent sequences of datasets. I got the  frame Trajectory using [this procedure](https://github.com/raulmur/ORB_SLAM/issues/28) can be compared to the ground truth of EuRoC dataset(download [here](http://www.cvlibs.net/datasets/kitti/eval_odometry.php)) using your evo package.

haidara@haidara-virtual-machine:~/catkin_ws/src/evo/evo/test/data$ evo_ape euroc V1_01_data.csv lsd_V101.txt -va --plot --save_results results/V.zip
--------------------------------------------------------------------------------
forcing trajectory alignment implicitly (EuRoC ground truth is in IMU frame)
--------------------------------------------------------------------------------
loaded 28712 stamps and poses from: V1_01_data.csv
[ERROR]
unhandled error in evo.main_ape
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 100, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_ape.py", line 254, in run
    args.t_offset,
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 228, in load_assoc_euroc_trajectories
    traj_est = read_tum_trajectory_file(est_file)
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 70, in read_tum_trajectory_file
    mat = np.array(csv_read_matrix(file_path, delim=" ", comment_str="#")).astype(float)
ValueError: invalid literal for float(): 0.0,0.0,0.0,0.0,0.0,0.0,0.0,1.0

[ERROR]
evo module evo.main_ape crashed - no logfile written (disabled)

haidela, 2017-11-13:
[lsd_V101.txt](https://github.com/MichaelGrupp/evo/files/1465617/lsd_V101.txt)

haidela, 2017-11-13:
That file is the file frametrajectory got by using  [this procedure](I'm using default parameters on KFUsageWegiht and KFDistWeight.
On calibration: "We recommend to use the ROS default parameter set (i.e., without undistortion), as undistortion of the pre-registered depth images is not trivial" - My interpretation is that as long as you don't use depth images then it should not be a problem to use rectified color images.
Scale and ATE calculation: See the small repository I uploaded [1]. I use a matlab script [2] and link to it in my script "ate_test.py". Steps: 1 To record keyframes I use "rostopic echo -p /lsd_slam/keyframes/camToWorld > lsd_camToWorld.txt" and "rostopic echo -p /lsd_slam/keyframes/time > lsd_time.txt". 2 process these with "python lsd_to_readable.py lsd_time.txt lsd_camToWorld.txt lsd.txt(<-the output text file)". 3. Process it all with "python ate_test.py lsd.txt gt.txt(e.g. rgbd_dataset_freiburg3_sitting_xyz-groundtruth.txt). This first uses "associate.py" to find corresponding timestamps, then it uses the matlab script "getScale.m" which again links to "absor.m"[2] to get the scale. Then it performs the ATE test with "evaluate_ate.py".
To link matlab and python, see [3]
I have included my test result form fr3/sit_xyz in [1] so you can play around.
There are som hardcoded parameters and all around lazy code in this as I thought I was only going to use it myself. You have to change the path in "ate_test.py" at least. Ask if there's any confusion.
Hope this helps

[1] https://github.com/bjornph/lsd_files
[2] http://www.mathworks.com/matlabcentral/fileexchange/26186-absolute-orientation-horn-s-method
[3] http://se.mathworks.com/help/matlab/matlab_external/install-the-matlab-engine-for-python.html#responsive_offcanvas)

haidela, 2017-11-13:
That file is the file frametrajectory got by using [this procedure](https://github.com/raulmur/ORB_SLAM/issues/28)

MichaelGrupp, 2017-11-13:
`evo_ape euroc` expects the "`est_file`" argument in the TUM format. Your lsd_V101.txt file is not exactly in this format - it has 8 entries per row but they are separated by commas, not space. I started a Wiki (still under construction) where you can find more infos about the formats [here](https://github.com/MichaelGrupp/evo/wiki/Formats).

I didn't read through your link with the "procedure", but I would recommend to just record the pose topic of LSD-SLAM to a bagfile (something like `rosbag record /lsd_slam/pose -O lsd_V101.bag` should work, I didn't check it). Then you can export it to TUM format using `evo_traj` with 
```evo_traj bag lsd_V101.bag --all_topics --save_as_tum```
like explained in the bottom of the linked Wiki page.

haidela, 2017-11-13:
Hello @MichaelGrupp Thank you very much for your quick reply and your valued assistance, I am first trying rosbag record /lsd_slam/pose -o /file.bag, see below

haidela, 2017-11-13:
haidara@haidara-virtual-machine:~/catkin_ws$ rosbag record /lsd_slam/pose -o '/media/haidara/IRM_CCSA_X6/MH_01_easy.bag'  
[ INFO] [1510574288.521546317]: Subscribing to /lsd_slam/pose
[ INFO] [1510574288.690217815]: Recording to /media/haidara/IRM_CCSA_X6/MH_01_easy_2017-11-13-19-58-08.bag.

haidela, 2017-11-13:
but it crashes, so I am trying that. I will try again

haidela, 2017-11-13:
I am trying and there is always a error
haidara@haidara-virtual-machine:~/catkin_ws/src/evo/evo/test/data$ evo_traj bag MH_01_easy_2017-11-13-19-53-48.bag --all_topics --save_as_tum 
--------------------------------------------------------------------------------
topicmanager initialized
[ERROR]
no geometry_msgs/PoseStamped topics found!
signal_shutdown [atexit]

MichaelGrupp, 2017-11-13:
Do a `rosbag info ...`on your bagfile - it looks like it has not recorded a `geometry_msgs/PoseStamped` topic. The `rosbag record /lsd_slam/pose ...` command was just a hint, I didn't check if it works - you have to do that yourself. Like I said in your last issue, I can't help you much with using some SLAM algorithm.

By the way: please put console output into a Markdown [fenced code block](https://help.github.com/articles/creating-and-highlighting-code-blocks/)
Otherwise the formatting will look weird. See also how I edited your last comments.

haidela, 2017-11-13:
@MichaelGrupp Thank you again for your generosity and I will continue to learn lsd_slam

closed 2017-11-13 by MichaelGrupp

cumtchenchang, 2017-12-18:
@haidela  
thank you for your question. Do you solved the problem.

905805952, 2018-01-17:
@cumtchenchang 
Excuse me,have you solved the problem?

cumtchenchang, 2018-01-17:
@905805952  solved. follow the step that @MichaelGrupp  said.