MichaelGrupp/evo #7: time offset should go before alignment

sume-cn, 2017-12-14:
### Bug / Error report

**Description:**

thank you for your great job, it helps a lot to my work

a possible bug found here:
main_traj.py 
line 218 do the alignment
line 242 do the time offset
the time offset block has no effect on my time shifted trajectory, 
so I change the two blocks, then it works

MichaelGrupp, 2017-12-14:
Thanks for reporting this, I don't really use the `t_offset` parameter so I wouldn't have noticed it ;) I just had a quick look it looks like you're right that this should come before alignment.

This is fixed by d2eb8da.

closed 2017-12-14 by MichaelGrupp

MichaelGrupp, 2017-12-14:
I also added the `--t_max_offset` parameter to `evo_traj`, so you can change the tolerance of the timestamp association like in the RPE or APE executables.

Just `git pull` or `pip install evo --upgrade` the new release to get the changes.

sume-cn, 2017-12-19:
I think you mean --t_max_diff, thanks!

MichaelGrupp, 2017-12-19:
Oops, yes! :D

commit d2eb8dae (2017-12-14): Fixes #7