MichaelGrupp/evo #8: RMSE in result

MichaelGrupp, 2017-12-25:
(Follow-up of #6)

@cumtchenchang
"Now more and more SLAM algorthms use RMSE to evelate the error, would you add RMSE in your code. I thank it will help more people."

MichaelGrupp, 2017-12-25:
@cumtchenchang
Can you explain where exactly you miss the RMSE? Because the RMSE is already implemented. If you run for example `evo_ape`, you will get output on the console in this format:
```
APE w.r.t. translation part (m)
(with SE(3) Umeyama alignment)

       max	0.024300
      mean	0.007492
    median	0.007415
       min	0.000350
      rmse	0.008119
       sse	0.143305
       std	0.003129
```
where `rmse` is the RMSE.

cumtchenchang, 2017-12-25:
sorry I do not pay more attention to that, It is my fault

MichaelGrupp, 2017-12-25:
No problem.

closed 2017-12-25 by MichaelGrupp