MichaelGrupp/evo #9: evo_res valueerror:shape of passed values is (2,1346),indices imply(2,1328)

wangshuailpp, 2017-12-29:
hello,
   I have already done evo_ape,so I got two results.But when I used evo_res to compare the two results.then the error happened.
thank you!

MichaelGrupp, 2017-12-29:
Thanks for reporting this - but first of all I need more data from your side to help you. Please use the issue template: https://github.com/MichaelGrupp/evo/blob/master/ISSUE_TEMPLATE.md (click on "Raw", copy the part under `### Bug / Error report
`, paste it here and answer the questions. That is really useful information for me.

In your case, I'm further interested if I can reproduce this error on my machine. So please upload the files you used here, if that's ok for you:
* trajectory files
* command for `evo_ape` that you used to create the results

I have already seen a similar error, but it was only once and I couldn't really reproduce it. Besides that, I wanted to refactor/rewrite parts of `evo_res` in the near future and for testing that your data could be very useful.

Thanks

wangshuailpp, 2017-12-29:
### Bug / Error report

**Description:**

**Command:**
```
# evo_ape euroc data.csv vins_result.txt -va --plot --save_results results/vins.zip
evo_ape euroc data.csv vins_result_lpp.txt -va --plot --save_results results/lpp.zip
evo_res results/*.zip -p --save_table results/table.csv
```

**Console output:**
```

#
[ERROR]
unhandled error in evo.main_res
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 93, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_res.py", line 126, in run
    raw_df = pd.concat([raw_df, new_raw_df], axis=1)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/reshape/concat.py", line 213, in concat
    return op.get_result()
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/reshape/concat.py", line 408, in get_result
    copy=self.copy)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 5207, in concatenate_block_managers
    return BlockManager(blocks, axes)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 3033, in __init__
    self._verify_integrity()
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 3244, in _verify_integrity
    construction_error(tot_items, block.shape[1:], self.axes)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 4608, in construction_error
    passed, implied))
ValueError: Shape of passed values is (2, 1346), indices imply (2, 1328)

[ERROR]
evo module evo.main_res crashed - no logfile written (disabled)

```

**Additional files:**
Please attach files if you can - for example trajectory files, images...
[lpp.zip](https://github.com/MichaelGrupp/evo/files/1593007/lpp.zip)
[vins.zip](https://github.com/MichaelGrupp/evo/files/1593008/vins.zip)

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: python2.7
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04 LST
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
[evo_config.txt.txt](https://github.com/MichaelGrupp/evo/files/1593025/evo_config.txt.txt)

```

# [evo_config.txt.txt](https://github.com/MichaelGrupp/evo/files/1593025/evo_config.txt.txt)

```

---

MichaelGrupp, 2017-12-29:
Ok thank you. Can you please also give me the files "vins_result.txt" and "vins_result_lpp.txt"? 
And the version number that you get if you run `evo pkg --version`.

Then I will have a look at what could be the problem.

wangshuailpp, 2017-12-30:
[Uploading vins_result_lpp.txt…]()
[vins_result.txt](https://github.com/MichaelGrupp/evo/files/1594196/vins_result.txt)
thank you for your help! there are two file you want,I have updated the version number of evo .

closed 2017-12-31 by MichaelGrupp

MichaelGrupp, 2017-12-31:
Ok, I found the reason for this problem: your trajectories have duplicate timestamps.

This is currently ignored by `evo_ape`, which means you can also get duplicate timestamps in your result file. In `evo_res`, duplicate indices will crash the `concat` function of the Pandas library and this error happens.

From my side, I added the following to fix this in commit https://github.com/MichaelGrupp/evo/commit/fb994bffe78a2804dcc1b22799b5e2e00817b4d0:
* `evo_res` now shows a warning if there are duplicate indices
* you can check if a trajectory has bad timestamps with `evo_traj ... --full_check`

Just `git pull` or run `pip install evo --upgrade` to get the new release.

But to fix the problem at its root, you should also check the code of your SLAM algorithm: the same timestamp multiple times in a trajectory makes no sense 😉

wangshuailpp, 2018-01-02:
Thank you very much!After the update I can run with "vins_result_lpp.txt" and "vins_result.txt" successfully.But when I run with others ,I get this problem again .You say that my trajectories have duplicate timestamps,But I can't find them.Could you point out for
 me? There are files I tested and the dataset I test is V1_01_easy.Thank you!
[CameraTrajectory.txt](https://github.com/MichaelGrupp/evo/files/1597052/CameraTrajectory.txt)
[vins_result.txt](https://github.com/MichaelGrupp/evo/files/1597053/vins_result.txt)
[vins_result_ws.txt](https://github.com/MichaelGrupp/evo/files/1597064/vins_result_ws.txt)

MichaelGrupp, 2018-01-02:
*"But when I run with others ,I get this problem again"*

What do you mean: Do you get the ValueError again? Or the new warning message? If your trajectories have duplicate timestamps, the warning message is expected behavior.

You can easily print the duplicated timestamps with this Bash command:
```bash
cat vins_result.txt | cut -d" " -f 1 | uniq -D
```
But you have to find out yourself *why* you have double timestamps.

wangshuailpp, 2018-01-03:
I mean that I get the ValueError again.Now I found the duplicate timestamps in "vins_result.txt" and erase these timestamps,then successful operation(plot results chat) happens with the waring "data lengths/indices are not consistent, plotting could make no sense". Thank you!

MichaelGrupp, 2018-01-03:
*"data lengths/indices are not consistent, plotting could make no sense"*

Don't worry, you can ignore this in most cases. This just means that the timestamps (or indices for KITTI) are not the same for every result. This just means that the "raw" value plot in the first tab could be not 100% correct (you can use the `--use_abs_time` flag to plot with absolute timestamps instead of relative timestamps). All other plots (histograms etc.) are fine. 

I agree that this message is confusing and it is not needed in most of the cases. I'll try to change this soon (issue #10). But right now you can just run `evo_res` with `--use_abs_time` and ignore the warning.

MichaelGrupp, 2018-01-13:
FYI: the "plotting could make no sense" message was changed in the new version. Also, `--use_abs_time` was removed because now absolute timestamps are the default.

Run this to upgrade:
```
sudo -H pip install evo --upgrade
```

commit fb994bff (2017-12-31): evo_res detects duplicate indices (resolves #9)