MichaelGrupp/evo #11: Install error on Ubuntu 14.04

YacongWang, 2018-01-04:
### Bug / Error report

**Description:**

When I install the evo on my OS, I have encountered the error.And I have tried several means to install the evo repo. But all failed.
**Command:**
```
# replace this line with the command you used
```
pip install evo --upgrade -i http://pypi.douban.com/simple
pip2 install evo --upgrade -i http://pypi.douban.com/simple
pip install evo --upgrade
sudo pip install evo --upgrade -i http://pypi.douban.com/simple
**Console output:**
```
 Downloading pyparsing-2.2.0-py2.py3-none-any.whl (56kB): 56kB downloaded
Installing collected packages: evo, matplotlib, scipy, pandas, seaborn, natsort, argcomplete, colorama, pygments, enum34, six, python-dateutil, backports.functools-lru-cache, subprocess32, pytz, cycler, pyparsing
  Found existing installation: evo v0.9.7-
    Uninstalling evo:
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 283, in run
    requirement_set.install(install_options, global_options, root=options.root_path)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1431, in install
    requirement.uninstall(auto_confirm=True)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 598, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1836, in remove
    renames(path, new_path)
  File "/usr/lib/python2.7/dist-packages/pip/util.py", line 295, in renames
    shutil.move(old, new)
  File "/usr/lib/python2.7/shutil.py", line 303, in move
    os.unlink(src)
OSError: [Errno 13] Permission denied: '/usr/local/bin/evo'

Storing debug log for failure in /home/vio/.pip/pip.log

---------------------------
After sudo* shell:

src/qhull_wrap.c: In function ‘delaunay_impl’:

src/qhull_wrap.c:114:5: error: unknown type name ‘QHULL_LIB_CHECK’

     QHULL_LIB_CHECK

     ^

src/qhull_wrap.c:117:5: error: conflicting types for ‘points’

     points = (coordT*)malloc(npoints*ndim*sizeof(coordT));

     ^

src/qhull_wrap.c:96:13: note: previous definition of ‘points’ was here

     coordT* points = NULL;

             ^

src/qhull_wrap.c:117:14: warning: initialization makes integer from pointer without a cast [enabled by default]

     points = (coordT*)malloc(npoints*ndim*sizeof(coordT));

              ^

src/qhull_wrap.c:118:16: warning: comparison between pointer and integer [enabled by default]

     if (points == NULL) {

                ^

src/qhull_wrap.c:134:15: error: subscripted value is neither array nor pointer nor vector

         points[2*i  ] = x[i] - x_mean;

               ^

src/qhull_wrap.c:135:15: error: subscripted value is neither array nor pointer nor vector

         points[2*i+1] = y[i] - y_mean;

               ^

src/qhull_wrap.c:157:29: warning: passing argument 3 of ‘qh_new_qhull’ makes pointer from integer without a cast [enabled by default]

                             "qhull d Qt Qbb Qc Qz", NULL, error_file);

                             ^

In file included from /usr/include/libqhull/qhull_a.h:27:0,

                 from src/qhull_wrap.c:10:

/usr/include/libqhull/libqhull.h:1002:9: note: expected ‘double *’ but argument is of type ‘int’

 int     qh_new_qhull(int dim, int numpoints, coordT *points, boolT ismalloc,

         ^

src/qhull_wrap.c:240:5: warning: passing argument 1 of ‘free’ makes pointer from integer without a cast [enabled by default]

     free(points);

     ^

In file included from /usr/include/python2.7/Python.h:42:0,

                 from src/qhull_wrap.c:8:

/usr/include/stdlib.h:483:13: note: expected ‘void *’ but argument is of type ‘int’

 extern void free (void *__ptr) __THROW;

             ^

src/qhull_wrap.c:259:5: warning: passing argument 1 of ‘free’ makes pointer from integer without a cast [enabled by default]

     free(points);

     ^

In file included from /usr/include/python2.7/Python.h:42:0,

                 from src/qhull_wrap.c:8:

/usr/include/stdlib.h:483:13: note: expected ‘void *’ but argument is of type ‘int’

 extern void free (void *__ptr) __THROW;

             ^

error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

  Rolling back uninstall of matplotlib
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip_build_root/matplotlib/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-rgYXYS-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip_build_root/matplotlib
Traceback (most recent call last):
  File "/usr/bin/pip2", line 9, in <module>
    load_entry_point('pip==1.5.4', 'console_scripts', 'pip2')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 185, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 29: ordinal not in range(128)

**Additional files:**
Please attach files if you can - for example trajectory files, images...

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 

Ubuntu 14.04
Python 2.7.6
---

MichaelGrupp, 2018-01-04:
Looks weird.
Normally, this should work fine on Ubuntu: `sudo -H pip install evo --upgrade`

After googling the error, I suspect this is an error caused by matplotlib:
* related issue: https://github.com/matplotlib/matplotlib/issues/8872
* pull request with a fix: https://github.com/matplotlib/matplotlib/pull/10074

You can check if the same error happens with: `sudo -H pip install matplotlib --upgrade`

As of today, the fix is not released yet (https://github.com/matplotlib/matplotlib/releases). Maybe you can try to do the same like the guy in [this comment](https://github.com/matplotlib/matplotlib/issues/8872#issuecomment-337799793) and upgrade this "qhull" library manually?

YacongWang, 2018-01-04:
Thanks Michael, following the upgrading "qhull" instruction in the matplotlib issues, and install matplotlib before install evo, I have solved that problem and installed the evo successfully. BTW, after matplotlib installation, I have encountered the error of "library dfftpack has Fortran sources but no Fortran compiler found", so I install the gfortran, after that the evo installation succeed. 
Looking forward to the bug fix of the matplotlib.

closed 2018-01-04 by MichaelGrupp