MichaelGrupp/evo #12: EuRoC dataset cannpt be used to evo

YacongWang, 2018-01-04:
### Bug / Error report

**Description:**

**Command:**
```
evo_ape euroc '/home/vio/trajectory_analysis/results of dataset/groundtruth/V102_groundtruth.csv' '/home/vio/trajectory_analysis/results of dataset/EuRoC/vins_V102.txt' -c '/home/vio/evo/doc/examples/config_ape.example.json' 
```

**Console output:**
```

vio@vio-wang:~$ evo_ape euroc '/home/vio/trajectory_analysis/results of dataset/groundtruth/V102_groundtruth.csv' '/home/vio/trajectory_analysis/results of dataset/EuRoC/vins_V102.txt' -c '/home/vio/evo/doc/examples/config_ape.example.json' 
--------------------------------------------------------------------------------
forcing trajectory alignment implicitly (EuRoC ground truth is in IMU frame)
--------------------------------------------------------------------------------
loaded 16702 stamps and poses from: /home/vio/trajectory_analysis/results of dataset/groundtruth/V102_groundtruth.csv
loaded 807 stamps and poses from: /home/vio/trajectory_analysis/results of dataset/EuRoC/vins_V102.txt
[ERROR]
unhandled error in evo.main_ape
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 93, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_ape.py", line 255, in run
    args.t_offset,
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 231, in load_assoc_euroc_trajectories
    ref_file, est_file)
  File "/usr/local/lib/python2.7/dist-packages/evo/core/sync.py", line 97, in associate_trajectories
    + " (s) and time offset: " + str(offset_2) + " (s)")
SyncException: found no matching timestamps between /home/vio/trajectory_analysis/results of dataset/groundtruth/V102_groundtruth.csv and /home/vio/trajectory_analysis/results of dataset/EuRoC/vins_V102.txt with max. time diff.: 100 (s) and time offset: 0.0 (s)

[ERROR]
evo module evo.main_ape crashed - no logfile written (disabled)

```

**Additional files:**
Please attach files if you can - for example trajectory files, images...

Please give also the following information:
* evo version number shown by `evo pkg --version`: evo v0.9.7
* Python version shown by `evo pkg --pyversion`: Python 2.7.6
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 14.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
{
    "logfile_enabled": false, 
    "logging_format": "%(message)s", 
    "plot_backend": "Qt4Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontsize": 12, 
    "plot_hideref": false, 
    "plot_info_text": false, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_format": "csv", 
    "table_export_transpose": true
}
[Uploading V102_groundtruth.csv…]()
[vins_V102.txt](https://github.com/MichaelGrupp/evo/files/1603700/vins_V102.txt)

MichaelGrupp, 2018-01-04:
Thanks for the detailed description and the files, this was very helpful to find the problem.

Looking at the lines in each of your files with `tail -n1 vins_102.txt V102_groundtruth.csv`:
```
==> vins_V102.txt <==
1403715609312143104 9.674000e-02 -7.041000e-02 7.675000e-02 8.166200e-01 -2.655000e-02 5.758600e-01 2.846000e-02

==> V102_groundtruth.csv <==
1403715608412143104,0.524978,1.987190,0.971496,0.159222,0.790116,-0.206917,0.554572,0.002414,0.010552,0.003042,-0.002162,0.020805,0.075824,-0.014726,0.105050,0.092967
```

* `V102_groundtruth.csv` has the correct "Euroc" format (timestamps in nanoseconds)
* the timestamps of `vins_V102.txt` are also in nanoseconds, but they have to be in seconds for the "TUM" format [(see here)](https://github.com/MichaelGrupp/evo/wiki/Formats#tum---tum-rgb-d-dataset-trajectory-format)

You need to divide the timestamps by 1e9, e.g. with a small script.

YacongWang, 2018-01-05:
Thanks Michael, can you provide a script to transform the vins_V102.txt, I don't succeed in transforming the format of vins_V102.txt.

MichaelGrupp, 2018-01-05:
Ok, normally I wouldn't write such things for you but this one is just 4 lines in Python if you already have evo installed:
```python
from evo.tools import file_interface
traj = file_interface.read_tum_trajectory_file("vins_V102.txt")
traj.timestamps = traj.timestamps / 1e9
file_interface.write_tum_trajectory_file("vins_V102_clean.txt", traj)
```

YacongWang, 2018-01-08:
Thanks Michael, This script works properly for me. That solved my problems.

closed 2018-01-08 by MichaelGrupp