MichaelGrupp/evo #13: something wrong with the evo_traj

YacongWang, 2018-01-05:
### Bug / Error report

**Description:**
something wrong with the evo_traj

**Command:**
```
evo_traj tum '/home/vio/evo/test/data/fr2_desk_ORB.txt' -p
```

**Console output:**
```

--------------------------------------------------------------------------------
name:	fr2_desk_ORB
infos:	2893 poses, 21.730m path length, 98.822s duration
[ERROR]
unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 93, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_traj.py", line 278, in run
    ax_traj = plot.prepare_axis(fig_traj, plot_mode)
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/plot.py", line 224, in prepare_axis
    ax = fig.add_subplot(subplot_arg, projection="3d", aspect="equal")
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/figure.py", line 1074, in add_subplot
    a = subplot_class_factory(projection_class)(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axes/_subplots.py", line 73, in __init__
    self._axes_class.__init__(self, fig, self.figbox, **kwargs)
  File "/usr/lib/pymodules/python2.7/mpl_toolkits/mplot3d/axes3d.py", line 91, in __init__
    *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axes/_base.py", line 541, in __init__
    self.cla()
  File "/usr/lib/pymodules/python2.7/mpl_toolkits/mplot3d/axes3d.py", line 1045, in cla
    Axes.cla(self)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axes/_base.py", line 1048, in cla
    self.grid(False)  # Disable grid on init to use rcParameter
  File "/usr/lib/pymodules/python2.7/mpl_toolkits/mplot3d/axes3d.py", line 1256, in grid
    self._draw_grid = maxes._string_to_bool(b)
AttributeError: 'module' object has no attribute '_string_to_bool'

[ERROR]
evo module evo.main_traj crashed - no logfile written (disabled)

```

**Additional files:**
Please attach files if you can - for example trajectory files, images...

Please give also the following information:
* evo version number shown by `evo pkg --version`: evo v0.9.7
* Python version shown by `evo pkg --pyversion`: 2.7.6
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 14.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```
{
    "logfile_enabled": false, 
    "logging_format": "%(message)s", 
    "plot_backend": "Qt4Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontsize": 12, 
    "plot_hideref": false, 
    "plot_info_text": false, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_format": "csv", 
    "table_export_transpose": true
}
```

MichaelGrupp, 2018-01-05:
Looks again like a matplotlib error. Please check the following:
* test if `evo_traj tum '/home/vio/evo/test/data/fr2_desk_ORB.txt' -p --plot_mode=xy` works
* if this works:
  * it's probably related to this issue: https://github.com/matplotlib/matplotlib/issues/3537/
  * what does `dpkg -l | grep matplotlib` print?

YacongWang, 2018-01-05:
Thanks Michael, it works with the --plot_mode=xy, and the print is:
,,,
vio@vio-wang:~$ dpkg -l | grep matplotlib
ii  python-matplotlib                                     1.3.1-1ubuntu5                                      amd64        Python based plotting system in a style similar to Matlab
ii  python-matplotlib-data                                1.3.1-1ubuntu5                                      all          Python based plotting system (data package)
,,,

YacongWang, 2018-01-05:
BTW, my matplotlib installation by pip , however qhull installation by .deb just as the above issue answer. Seems that solution in the matplotlib/matplotlib#3537 don't work for me. So I don't know what to do?

MichaelGrupp, 2018-01-05:
Then it's the error that is discussed in the matplotlib issue. You have conflicting versions of matplotlib installed:
* version 1.3.1 installed by `apt` as shown by the `dpkg` call
* the newer version installed by `pip`

By the way: I use Ubuntu 16.04 with matplotlib 1.5.1 from `apt` and 2.0.1 from `pip` without problems. Unfortunately, there is no 1.5.1 package for Ubuntu 14.04.

MichaelGrupp, 2018-01-05:
*"Seems that solution in the matplotlib/matplotlib#3537 don't work for me"*

Which solution? Did you try to remove the Ubuntu packages with:
```
sudo apt remove python-matplotlib python-matplotlib-data
```

YacongWang, 2018-01-08:
Thanks Michael. Following the steps like: 
,,,
sudo apt remove python-matplotlib python-matplotlib-data
sudo -H pip uninstall matplotlib
sudo -H pip install matplotlib
,,,
I have solved the conflict. Thanks for your immediate answer to help me.

MichaelGrupp, 2018-01-08:
Great!

closed 2018-01-08 by MichaelGrupp