MichaelGrupp/evo #14: transform the background color of plotting figures

YacongWang, 2018-01-08:
### Feature request

Hello Michael, I want to transform the background color of figures that are plotted by ```--plot --plot_mode``` to white just like MATLAB, because when the figure print on paper, there is a lot blur for the lines with the impact of background color. What should I do to achieve my goal?

MichaelGrupp, 2018-01-08:
I use [Seaborn](http://seaborn.pydata.org/) for plot styling. There are different "seaborn styles" that will change the background of your figure: http://seaborn.pydata.org/tutorial/aesthetics.html#seaborn-figure-styles

There is a config parameter in evo for exactly that. For example, if you want to change the style to `white`or `whitegrid`, just run:
```
evo_config set evo_config set plot_seaborn_style whitegrid
```

`evo_config show` will explain you some other parameters you can play around with. Changing some of the `plot_*` is probably what you want to do before printing.

MichaelGrupp, 2018-01-08:
And if you mess up your config, `evo_config reset` will reset everything to the default values again. Have fun :grin: 

For everything else: https://github.com/MichaelGrupp/evo/wiki/Configuration

MichaelGrupp, 2018-01-09:
Can I close that issue?

YacongWang, 2018-01-10:
Thanks Michael, that works. It solved my problems.

closed 2018-01-10 by MichaelGrupp