MichaelGrupp/evo #16: Questions about rosbag of geometry_msgs/PoseStamped

YacongWang, 2018-01-10:
### Feature request

When using ``` rosbag record ```to record the ```geometry_msgs/PoseStamped ```topic, it used system time and not the timestamps of dataset, so that bag cannot align with the groundtruth. It seems the ```rosbag record``` dosen't work to save timestamps, positon and orientation, and I want to know what should I do to use the feature of bag format which can be used to evo repo. Thank you Michael.

MichaelGrupp, 2018-01-10:
Are you playing the dataset from a bagfile? If yes, you need to set the ROS param `use_sim_time` and specify the `--clock` flag for `rosbag play` in order to use the bag's clock. http://wiki.ros.org/Clock

```bash
rosparam set use_sim_time true  # (or set in launch file if you use one)
rosbag play <your bag> --clock
```

MichaelGrupp, 2018-01-12:
Closing this, feel free to re-open it still doesn't work.

closed 2018-01-12 by MichaelGrupp

YacongWang, 2018-01-14:
Thanks for your help. That works for my algorithm.