MichaelGrupp/evo #17: Error about evo_res

YacongWang, 2018-01-14:
### Bug / Error report

**Description:**
I have encountered the error
[results_vins_rpe.tar.gz](https://github.com/MichaelGrupp/evo/files/1629741/results_vins_rpe.tar.gz)
[results_vins_rpe.tar.gz](https://github.com/MichaelGrupp/evo/files/1629742/results_vins_rpe.tar.gz)

 when I compare the results from rpe w.r.t. translation part.

**Command:**
```
evo_res vins_MH04.zip vins_MH05.zip  vins_V102.zip  vins_V202.zip  --save_table rpe_table_ms_vi.csv --plot
```

**Console output:**
```
[WARNING] duplicate indices in error array of vins_MH04 - keeping only first occurrence of duplicates
[WARNING] duplicate indices in error array of vins_MH05 - keeping only first occurrence of duplicates
[ERROR] unhandled error in evo.main_res
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 93, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_res.py", line 129, in run
    error_df = pd.concat([error_df, new_error_df], axis=1)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/reshape/concat.py", line 213, in concat
    return op.get_result()
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/reshape/concat.py", line 408, in get_result
    copy=self.copy)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 5207, in concatenate_block_managers
    return BlockManager(blocks, axes)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 3033, in __init__
    self._verify_integrity()
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 3244, in _verify_integrity
    construction_error(tot_items, block.shape[1:], self.axes)
  File "/usr/local/lib/python2.7/dist-packages/pandas/core/internals.py", line 4608, in construction_error
    passed, implied))
ValueError: Shape of passed values is (2, 2076), indices imply (2, 2068)

[ERROR] evo module evo.main_res crashed - no logfile written (disabled)

```

**Additional files:**
Please attach files if you can - for example trajectory files, images...

Please give also the following information:
* evo version number shown by `evo pkg --version`: v0.9.8
* Python version shown by `evo pkg --pyversion`: 2.7.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```
{
    "logfile_enabled": false, 
    "logging_format": "%(message)s", 
    "plot_backend": "Qt4Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontsize": 12, 
    "plot_hideref": false, 
    "plot_info_text": false, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_seaborn_style": "white", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_format": "csv", 
    "table_export_transpose": true
}
```
[results_vins_rpe.tar.gz](https://github.com/MichaelGrupp/evo/files/1629743/results_vins_rpe.tar.gz)

MichaelGrupp, 2018-01-15:
Same issue as in #9: your data has duplicate timestamps, which shouldn't happen. There is probably a bug in your SLAM system - the other guy in #9 also used "vins".

From my side: 
* fixed the logic that checks for duplicates a bit, so that it can also handle your case which was a bit different (https://github.com/MichaelGrupp/evo/commit/cbff9a535cb56bfcd3b469e6c2a1d2636e016549), now in version `v0.9.8.1`
* use [this script](https://github.com/MichaelGrupp/evo/blob/master/contrib/print_duplicate_timestamps.sh) to detect duplicate stamps
* fix your SLAM system

MichaelGrupp, 2018-01-15:
Edit: I had to [fix the fix](https://github.com/MichaelGrupp/evo/commit/edae619158cbd2e8a38a24a63445ec307db1ee0e) ;) Otherwise you always get the warning even if there are no duplicates. Please use `v0.9.8.2` - sorry for the inconvenience.

YacongWang, 2018-01-16:
Thanks for your update. That works for my problem.

closed 2018-01-16 by MichaelGrupp