MichaelGrupp/evo #18: Can't install the evo

izengpan, 2018-01-27:
### Bug / Error report

**Description:**

I have followed the two kinds of installation steps, but fail to install the evo.
**Command:**
```
#Just the lines you have shared.
```

**Console output:**
```

#  Can't roll back matplotlib; was not uninstalled
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip_build_root/matplotlib/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-KqvzVP-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip_build_root/matplotlib
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.4', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 235, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 29: ordinal not in range(128)

```

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

#  Ubuntu 14.04

```

MichaelGrupp, 2018-01-27:
Matplotlib is a bit annoying on Ubuntu 14.04. Can you have a look at issue #13 and try the uninstall commands from [this comment](https://github.com/MichaelGrupp/evo/issues/13#issuecomment-355555444)?

izengpan, 2018-01-28:
I follow the three steps:
sudo apt remove python-matplotlib python-matplotlib-data
sudo -H pip uninstall matplotlib
sudo -H pip install matplotlib

The first and third step works, while the second step can not work.
And then, I ran the code:  pip install evo --upgrade
There is new mistake, and I don't know how to handle this:
 error: library dfftpack has Fortran sources but no Fortran compiler found

MichaelGrupp, 2018-01-28:
There's something broken or incompatible. Have a look at what other people do, for example on Stack Overflow: https://stackoverflow.com/questions/29586487/still-cant-install-scipy-due-to-missing-fortran-compiler-after-brew-install-gcc

Other than that, I can't help much with this issue.

closed 2018-01-28 by MichaelGrupp

YacongWang, 2018-02-02:
@izengpan  You should run  ```sudo apt-get install gfortran```.