MichaelGrupp/evo #20: Feature Request:  ```evo_ape```

YacongWang, 2018-02-05:
Feature Request:
I want to compare several trajectories of different algorithms with the same groundtruth( the same scene). But the ```evo_ape``` just can compare one pair of groundtruth and estimation trajectories. Can you add a feature that like ```evo_traj groundtruth estimation_traj1 estimation_traj2 estimation_traj3 ... ``` to compare several estimation trajectories from different algorithms with the one groundtruth. And plot the ape error curve in one window.
Thanks for your help.

MichaelGrupp, 2018-02-05:
What would be the advantage over running this:
```bash
evo_ape tum groundtruth estimation_traj_1 --save_results 1.zip
evo_ape tum groundtruth estimation_traj_2 --save_results 2.zip
# ...
evo_ape tum groundtruth estimation_traj_n --save_results n.zip

evo_res *.zip ---plot
```

This can be easily automated for example with a for loop in bash script if you want to execute it with a single command.

There are several points why I wouldn't like this in the `evo_*pe` executables:
* Added complexity in `evo_*pe` - in the code as well as in usability. One example: how would you handle trajectory-specific options like `--t_offset`? Also the plot with the colormap: overlaying multiple of those would look weird.
* By design, the comparison of results is a separate step handled by `evo_res`. It is designed to handle `Result` [objects](https://github.com/MichaelGrupp/evo/blob/master/evo/core/result.py) from `.zip` files independent of how they were generated. I prefer this separation ("do one thing and do it well") over having everything in a single executable that is hard to maintain.
* `evo_traj` is a different case - it doesn't generate "results" in the sense of the metrics. Its purpose is to do stuff multiple trajectories.

YacongWang, 2018-02-05:
Thanks Michael. I want to plot error curves like [this](https://github.com/MichaelGrupp/evo/blob/master/doc/assets/res_raw.png), what's the configuration to be set? And can I plot the error of x, y, and z-axis like the image below? 
[image]

MichaelGrupp, 2018-02-05:
The subplot at the bottom is already like the one you would get with `evo_res`. By the way, I think your image has wrong labels - "distance [m]" doesn't make sense here? Should be time I think.

> can I plot the error of x, y, and z-axis like the image below?

Currently not. This would be a new option for `--pose_relation`, and I'm not sure if or when I would integrate something like that. If you need it urgently, you have to code something yourself. You can have a look at the [custom_app.py](https://github.com/MichaelGrupp/evo/blob/master/doc/examples/custom_app.py) example to see how you can make a custom script.

For example, I changed it here to calculate the x,y,z differences as a numpy matrix:

```python
#!/usr/bin/env python

from __future__ import print_function

print("loading required evo modules")
from evo.core import trajectory, sync, metrics
from evo.tools import file_interface

print("loading trajectories")
traj_ref = file_interface.read_tum_trajectory_file("../../test/data/fr2_desk_groundtruth.txt")
traj_est = file_interface.read_tum_trajectory_file("../../test/data/fr2_desk_ORB.txt")

print("registering and aligning trajectories")
traj_ref, traj_est = sync.associate_trajectories(traj_ref, traj_est)
traj_est = trajectory.align_trajectory(traj_est, traj_ref, correct_scale=False)

print("calculating xyz differences")
import numpy as np
diff = np.abs(traj_ref.positions_xyz - traj_est.positions_xyz)
print(diff)

# add plotting code here
import matplotlib.pyplot as plt
```

You can add matplotlib plotting code to that.

YacongWang, 2018-02-05:
Thanks Michael. And something wrong with my ```evo_res```, the raw ape image like that.
[image]
What's wrong with my evo software?

YacongWang, 2018-02-05:
Thanks Michael. I have tried to implement the feature. But, there is something struck me.
```
 x_idx, y_idx, z_idx = plot_mode_to_idx(plot_mode)
print("calculating xyz differences")
import numpy as np
diff = np.abs(traj_ref.positions_xyz - traj_est.positions_xyz)
print(diff[:,x_idx])
```
How should I call the ```plot_mode```?

YacongWang, 2018-02-26:
Thanks Michael. I have tried to implement the feature that plotting the error of x, y, and z-axis individually, but I encountered some issues. The py doc is uploaded below.

[ape_xyz.zip](https://github.com/MichaelGrupp/evo/files/1757713/ape_xyz.zip)

MichaelGrupp, 2018-02-26:
What are the issues?

YacongWang, 2018-02-26:
When I compare three trajectories including groundtruth, the output is
```
python ape_xyz.py 
loading required evo modules
loading trajectories
registering and aligning trajectories
Traceback (most recent call last):
  File "ape_xyz.py", line 32, in <module>
    ape_metric1.process_data(data1)
  File "/usr/local/lib/python2.7/dist-packages/evo/core/metrics.py", line 471, in process_data
    raise MetricsException("trajectories must have same number of poses")
evo.core.metrics.MetricsException: trajectories must have same number of poses
```

[ape_xyz_2.zip](https://github.com/MichaelGrupp/evo/files/1759092/ape_xyz_2.zip)

MichaelGrupp, 2018-02-26:
Ah, I see the problem :smile:  I just ran a quick debugging it with pudb - the problem is that you run the `sync.associate(...)` function twice with the same reference trajectory. The associate function synchronizes the trajectories so that they have the same number of matching poses.

So this part is totally fine to associate `traj_ref` and `traj_est1`:
```python
traj_ref, traj_est1 = sync.associate_trajectories(traj_ref, traj_est1)
traj_est1 = trajectory.align_trajectory(traj_est1, traj_ref, correct_scale=False)
```
However this part...
```python
traj_ref, traj_est2 = sync.associate_trajectories(traj_ref, traj_est2)
traj_est2 = trajectory.align_trajectory(traj_est2, traj_ref, correct_scale=False)
```
associates `traj_ref` to `traj_est2`. The number of poses of `traj_ref` changes again, which leads to the error because it doesn't match your first trajectory anymore.

So to solve your problem you need two separate reference trajectories (e.g. "`traj_ref1`" and "`traj_ref2`". Either load the ground truth file twice or use the [deepcopy function of Python](https://docs.python.org/2/library/copy.html) to create a copy of the ground truth before the `sync.associate(...)` calls.

YacongWang, 2018-02-27:
Thanks, Michael. After using the ground truth file twice, I have plotted the error curves successfully. But I also have another problem that the two error curves don't match and the images are uploaded below. Just like the above Fig. 12, the error curves of x,y and z-axes should have the same tendency.
What else should I add in the code?
[image]
[image]
[image]

[ape_xyz.zip](https://github.com/MichaelGrupp/evo/files/1761034/ape_xyz.zip)

YacongWang, 2018-02-27:
And when I want to compare the rotation angle like pitch, raw, and roll, what command should be added to replace the  ```diff1 = np.abs(traj_ref1.positions_xyz - traj_est1.positions_xyz)```.

MichaelGrupp, 2018-02-27:
Plot the values over time (`traj.timestamps`), not their array index. You don't need to match these 2 trajectories.

YacongWang, 2018-02-27:
Thanks. Can you tell me the plot command using the ```traj.timestamps```? And I use the following command to replace the ```diff1 = np.abs(traj_ref1.positions_xyz - traj_est1.positions_xyz)``` to obtain the error of rotation angle like pitch, raw, and roll. Is right?
```
diff1 = np.abs(traj_ref.orientations_euler - traj_est1.orientations_euler)
```

MichaelGrupp, 2018-02-27:
Since you're using `plot.error_array(...)` function, you can set the `x_array` parameter to the timestamps of the respective "diff" array. Have a look at the doc-string of this function [here](https://github.com/MichaelGrupp/evo/blob/6ccc62e41267042acf39e1e585dfdf6b74f79e30/evo/tools/plot.py#L430). The 

The `orientations_euler` are roll, pitch, yaw in x-y-z convention - [code](https://github.com/MichaelGrupp/evo/blob/6ccc62e41267042acf39e1e585dfdf6b74f79e30/evo/core/trajectory.py#L91).

YacongWang, 2018-02-27:
Thanks Michael. After using the code
```
if isinstance(traj_est1, trajectory.PoseTrajectory3D):
        seconds_from_start1 = [t - traj_est1.timestamps[0] for t in traj_est1.timestamps]
else:
        seconds_from_start1 = None
plot.error_array(fig_1, diff1_x, x_array=seconds_from_start1, color='grey',
                 name="msckf", title=str(ape_metric1), xlabel="time", linestyle="-")
```
I plot the error curves successfully.
```
diff1 = np.abs(traj_ref1.orientations_euler - traj_est1.orientations_euler)
diff1_yaw = diff1[:,0]#yaw-z
diff1_pitch = diff1[:,1]#pitch-y
diff1_roll = diff1[:,2]#roll-x
```
The ```diff1_yaw, diff1_pitch and diff1_roll``` is right?

MichaelGrupp, 2018-02-27:
It's up to you to define the axes for "roll", "pitch" and "yaw". In mobile robotics, the rotation around z is called yaw, however in cameras z regularly points "forward" (roll). The x, y, z order is 0, 1, 2 though, so your comments are wrong.

YacongWang, 2018-02-27:
Thanks Michael.
When I run the code, I have the warning
```
loading plot modules
plotting
/usr/local/lib/python2.7/dist-packages/matplotlib/cbook/deprecation.py:106: 
MatplotlibDeprecationWarning: Adding an axes using the same arguments as a previous axes currently reuses the earlier instance.  
In a future version, a new instance will always be created and returned.  
Meanwhile, this warning can be suppressed, and the future behavior ensured, by passing a unique label to each axes instance.
  warnings.warn(message, mplDeprecation, stacklevel=1)
```
What else should I do to avoid the warnings?
[ape_xyz.py.zip](https://github.com/MichaelGrupp/evo/files/1762715/ape_xyz.py.zip)

MichaelGrupp, 2018-02-27:
No idea. Debug it or ignore it.

closed 2018-02-27 by MichaelGrupp

cumtchenchang, 2018-05-15:
@YacongWang  Can you share the final successful code？