MichaelGrupp/evo #23: BUG about rpy of evo_traj

YacongWang, 2018-02-27:
### Bug / Error report

**Description:**

  The rotation angle like yaw of ground truth cannot correspond to the estimation.
[image]

**Command:**
```
evo_traj tum --ref='/home/vio/evo-master/test/data/MH05_groundtruth.txt' '/home/vio/evo-master/test/data/msckf_MH05.txt' -a -p
```

**Console output:**
```
name:	msckf_MH05
infos:	2221 poses, 101.631m path length, 111000000000.000s duration
--------------------------------------------------------------------------------
name:	MH05_groundtruth
infos:	22212 poses, 97.593m path length, 111055000064.000s duration

```

**Additional files:**

[evo_traj_rpy.zip](https://github.com/MichaelGrupp/evo/files/1763049/evo_traj_rpy.zip)

Please give also the following information:
* evo version number shown by `evo pkg --version`: v0.9.8.4
* Python version shown by `evo pkg --pyversion`: 2.7.12
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* did you change the source code? (yes / no): no
* output of `evo_config show --brief --no_color`: 
```
{
    "logfile_enabled": false, 
    "logging_format": "%(message)s", 
    "plot_backend": "Qt4Agg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontsize": 12, 
    "plot_hideref": false, 
    "plot_info_text": false, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}

```

MichaelGrupp, 2018-02-28:
This looks like the euroc dataset, right? Keep in mind that the ground truth trajectories of this dataset are in the IMU/body frame of the quadcopter, while the visual SLAM trajectories are usually defined in the left camera. So a rotated frame of reference is expected behavior also after point-to-point alignment (what evo does). You would need to properly transform the groundtruth from the IMU to the camera frame to allow such comparisons - have a look at the documentation of the dataset you might find this transformation there.

MichaelGrupp, 2018-02-28:
In general, I'd recommend to use the RPE metrics for such comparisons. There you don't have to care so much about the absolute frame of reference:
```bash
evo_rpe tum MH05_groundtruth.txt msckf_MH05.txt -a --pose_relation angle_deg -p
```

YacongWang, 2018-02-28:
Thanks Michael. Can you add such transformation to EuRoC ground truth by using such option ```--euroc_groundtruth``` when using ```--ref=euroc_groundtruth.txt```?

YacongWang, 2018-02-28:
And maybe we can use the ```traj_ref1.orientations_euler - traj_est1.orientations_euler``` to compare the rotation angle of axis for IMU-Camera has the same rotation angle.

MichaelGrupp, 2018-02-28:
This a very special requirement that doesn't suit the general use of `evo_traj`. But you can already apply transformations (see [here](https://github.com/MichaelGrupp/evo/wiki/evo_traj#alignment-and-transformation)) to trajectories if you need to. For example, you could have a pre-processing step that transforms your groundtruth and saves it (`--save_as_tum`).

Again: unless the frame of reference is the same, you have to use `evo_rpe`. This is very stable as it uses the logarithmic map to calculate the angle.

When I have time I'll change the behavior of the metrics, but not `evo_traj`: see https://github.com/MichaelGrupp/evo/issues/24

YacongWang, 2018-03-01:
Following your instruction, I still have wrong results. Maybe the reason is that the quaternion format of euroc ground truth(```timestamp x y z  q_w q_x q_y q_z```) is different from the ```tum``` format(```timestamp x y z q_x q_y q_z q_w```). When using ```evo_traj```, there don't have option ```--ref euroc_groundtruth.csv``` like ```evo_ape```.

MichaelGrupp, 2018-03-01:
`--transform_right` is probably what you need since you want to transform from the local frame.

YacongWang, 2018-03-01:
I have noticed that output ```111000000000.000s duration```, but the timestamp in the msckf_MH05.txt is corresponding to the ground truth. And the data.tum obtained from ```evo_traj euroc data.csv --save_as_tum``` cannot have the same timestamp with the original file-data.csv. Can you solve this problem? The source data and results are uploaded above.

MichaelGrupp, 2018-03-01:
> Can you solve this problem? The source data and results are uploaded above.

Come on - you had the same issues already here: https://github.com/MichaelGrupp/evo/issues/12

Easy to solve with [the timestamp script](https://github.com/MichaelGrupp/evo/blob/master/contrib/multiply_timestamps.py)

MichaelGrupp, 2018-03-01:
Closing as the initial problem is now known and there's nothing to do from my side.

closed 2018-03-01 by MichaelGrupp

YoungHaulk, 2023-05-27:
Hi, I also encountered the same problem. The xyz diagram of the trajectory is corresponding, but the roll, yaw and pitch angles are not aligned. How did you solve this problem?