MichaelGrupp/evo #27: Windows support?

alessandroTorresani, 2018-03-06:
Hi,
I am trying to intall evo on Windows 10 - Anaconda Python 3.6.
During installation I got and error about "pwd" module.

Any help?

Thank you!

MichaelGrupp, 2018-03-06:
Hey,

Windows support is something that is not yet tested from my side. But I know of some people who started working on that, the fork is here: https://github.com/paul-michalik/evo

The owner said that they will do a pull request once it is ready. Until this is done and merged in here: maybe you can try if their fork already works for you?

MichaelGrupp, 2018-03-06:
There are also some steps in the README: https://github.com/paul-michalik/evo#instalation-in-anaconda-environment 

You can try this

MichaelGrupp, 2018-04-22:
I fixed the Windows installation such that it builds on AppVeyor, see this commit: https://github.com/MichaelGrupp/evo/commit/9e0ed0aa19e8f254d7b5c88b702c7333ca5a6354

It should be possible to install the master branch / pypi release on Windows now.
Let me know if there are still problems.

closed 2018-04-22 by MichaelGrupp

alessandroTorresani, 2018-05-07:
Thank you very much and sorry for my late reply.
Your library helped me a lot.

MichaelGrupp, 2018-05-07:
No problem. Great to hear that!