MichaelGrupp/evo #31: [EuRoC Dataset] invalid literal for float()

andywangxb, 2018-03-13:
### [ERROR] unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/ubuntuandy/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/entry_points.py", line 93, in launch
    main_module.run(args)
  File "/home/ubuntuandy/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/main_traj.py", line 187, in run
    trajectories.append((csv_file, file_interface.read_euroc_csv_trajectory(csv_file)))
  File "/home/ubuntuandy/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/tools/file_interface.py", line 144, in read_euroc_csv_trajectory
    mat = np.array(csv_read_matrix(file_path, delim=",", comment_str="#")).astype(float)
#### ValueError: invalid literal for float(): 1.403715529112143517e+09 -6.151000000000000217e-02 4.837999999999999939e-02 1.771199999999999997e-01 8.132099999999999884e-01 -2.730000000000000135e-02 5.806599999999999540e-01 2.778999999999999873e-0

MichaelGrupp, 2018-03-13:
Looks like you use the wrong format, see the Wiki page on formats.

And please follow the issue template (that you didn't use) or provide at least the command and your data. Otherwise your report is useless for helping you.

MichaelGrupp, 2018-03-16:
Closing this as answered, feel free to provide more information if the problem still occurs.

closed 2018-03-16 by MichaelGrupp

andywangxb, 2018-03-20:
### Problem fixed
addressed problem by converting EuRoC data format to TUM data format.
#### find groundtruth data in *sequence folder*
`.../mav0/state_groundtruth_estimate0/data.csv`
#### Command
`evo_traj euroc data.csv --save_as_tum` 

details please check [here](https://github.com/MichaelGrupp/evo/wiki/Formats#saving--exporting-to-other-formats)
#### data evaluation
`evo_traj tum CameraTrajectory.txt --ref=data.tum -a -p`

MichaelGrupp, 2018-03-20:
Yes, for `evo_traj`all trajectories have to be in the same specified format.