MichaelGrupp/evo #33: signal_shutdown [atexit]

gongyue666, 2018-03-20:
### Bug / Error report

**Description:**

When I run evo_traj bag ROS_example.bag groundtruth ORB-SLAM S-PTAM in /eva/test/data/.This log always show , and I get no result then.
**Command:**
```
# replace this line with the command you used
```

**Console output:**
```

# remove this line and paste your console output HERE

```

**Additional files:**
Please attach files if you can - for example trajectory files, images...

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```

# remove this line and paste your config HERE

```

MichaelGrupp, 2018-03-20:
Please edit your comment and fill out all the information of the issue template.
Because `evo_traj bag ROS_example.bag groundtruth ORB-SLAM S-PTAM` works perfectly fine for me.

MichaelGrupp, 2018-03-23:
Closing this for now until more information is given.

By the way: the `signal_shutdown[atexit]` message was normal in the last versions, it's from the ROS modules. I changed the logging behavior in the newest version so that unimportant logs like these are not shown anymore. You can upgrade the package as described here: https://github.com/MichaelGrupp/evo#installation

closed 2018-03-23 by MichaelGrupp