MichaelGrupp/evo #41: Can you give an interface to set min and max value of evo_ape's colorbar

HeYijia, 2018-05-08:
Hi, dear author, 

Firstly, thank you very much to share this code. It saved me a lot of time in my research work.

I have a little suggestion. When we use evo_ape to evaluating method，the min and max value of the colorbar are auto selected so that different method has a different colorbar. But, I think if we use the same colorbar we can compare the difference between methods and ground truth more intuitively.

So, can you give an interface to set min and max value of evo_ape's colorbar? Thanks in advance.

MichaelGrupp, 2018-05-08:
Hi,

I agree that this would be a very useful addition. Currently, the range is simply set from the minimum to the maximum error value. I see two options that would be useful:

* user can specify a fixed range interval `[min, max]`
  * allows consistent comparisons, like you said
* user can specify a percentile of the error distribution as the `max` value
  * allows to ignore outliers

---
Kind of related to this topic:

A distribution plot next to the colorbar would be also nice, like it's done for example in CloudCompare: http://www.danielgm.net/cc/images/snapshot2_small.jpg
---

I will have a look at how this could be implemented the best way soon.

WongRZ, 2018-05-21:
Dear author, is this addition implemented?

MichaelGrupp, 2018-05-21:
Couldn't allocate time yet to do this. But I'll notify you here once I did.

closed 2018-05-23 by MichaelGrupp

MichaelGrupp, 2018-05-23:
This was added in the new release, see the linked pull request.

You can upgrade as described [here](https://github.com/MichaelGrupp/evo#installation).

PR #51 (2018-05-23): Add CLI options to set colormap bounds in evo_{ape, rpe}. 
--plot_colormap_max
--plot_colormap_min
--plot_colormap_max_percentile

Closes #41