MichaelGrupp/evo #47: Weird result with EuRoC dataset

HenningLuo, 2018-05-17:
### Bug / Error report

**Description:**
Hi, I was running trajectory generated by ORB-SLAM2 using EuRoC stereo dataset V1_01_easy, both the file under evaluation and the groundtruth file are in TUM format. However, the result seems a bit weird. Is it because that I did something incorrectly or that it is supposed to be like this? Many thanks.

**Command:**
The groundtruth file I used was 
`mav0/state_groundtruth_estimate0/data.csv`
and I transformed it to TUM format using
`evo_traj euroc data.csv 	--save_as_tum`
the command I typed to evaluate it was
`evo_traj tum -s -p -v -f --ref=data.tum --plot_mode=xz stereo_orb_result.txt`

**Additional files:**
[image]

MichaelGrupp, 2018-05-17:
The ground truth of the EuRoC dataset is not located in the camera coordinate frame, that is probably causing the offset.

In this case it's useful to align the SLAM trajectory to the ground truth with `--align` or `-a`. See also [here](https://github.com/MichaelGrupp/evo/wiki/evo_traj#alignment-and-transformation).

HenningLuo, 2018-05-18:
It worked! Thanks so much for your help.

closed 2018-05-18 by HenningLuo

mengqingyu123, 2018-06-27:
I'm sorry. When I use the -a command to align the trajectory, the attitude Angle is still not aligned. What's going on? Thank you very much!