MichaelGrupp/evo #48: In evo_ape, the result from full trajectory of EuRoC dataset is different from part of it!

WongRZ, 2018-05-18:
Hi, dear author,
Firstly, thank you very much to share this code. It helped me a lot.
I have a problem. when i use evo_ape to evaluate a result from full trajectory of EuRoC V202 dataset and  a part of the same trajectory, they both have the max error, but one is 0.670081 and another is 0.646771. these two max errors should be same one because they came from the same trajectory, but in this case ,they are different.

WongRZ, 2018-05-18:
May be you can test in your trajectory data

MichaelGrupp, 2018-05-18:
Please provide all data and steps how to reproduce (the [issue template](https://github.com/MichaelGrupp/evo/blob/master/ISSUE_TEMPLATE.md) is your friend :wink: ).

WongRZ, 2018-05-19:
And the groundtruth file called V202_groundtruth.csv is "data.csv" from folder named "state_groundtruth_estimate0" of EuRoC MAV ”Vicon Room 2 02“ dataset.

WongRZ, 2018-05-19:
Thanks a lot, dear author. I'm probably getting it. So if I want to compare two trajectories, their length should be the same?

MichaelGrupp, 2018-05-19:
No. But if you cut parts of a trajectory off, you can't expect the result to be the same.

(of course it usually _makes sense_ to compare trajectories of approximately the same length, but that's a different story)

closed 2018-05-19 by MichaelGrupp