MichaelGrupp/evo #50: cannot plot!

zhuqingzhang, 2018-05-20:
### Bug / Error report

 i have successfully installed evo, and i did some tests using it. there is something wrong with ploting

when i use the command:
`evo_traj kitti KITTI_00_ORB.txt`

it loaded the txt successfully
[image]

but
when i wanted to plot and use the commond:
`evo_traj kitti KITTI_00_ORB.txt --plot`
it shew that  **Segmentation fault(core dumped)**

[image]

And i have tried other command like **evo_ape**. But as long as i used **--plot** ,there was error
ANYONE KNOW HOW TO FIX IT?
```

MichaelGrupp, 2018-05-21:
* Check if basic matplotlib works for you. E.g. try some of those examples: https://matplotlib.org/tutorials/index.html - if this doesn't work, it's a matplotlib issue.
* If the above works, please provide more information on your setup using the [issue template](https://github.com/MichaelGrupp/evo/blob/master/ISSUE_TEMPLATE.md).

MichaelGrupp, 2018-05-21:
Side note: why do you run it as `root`??

MichaelGrupp, 2018-05-25:
Closing for inactivity. Reopen if you can provide more information.

closed 2018-05-25 by MichaelGrupp