MichaelGrupp/evo #55: new-line character seen in unquoted field

xukuanHIT, 2018-06-08:
Hi, thank you for your great work.
  I meet an error when I use evo with euroc dataset, the error is as follows,

Forcing trajectory alignment implicitly (EuRoC ground truth is in IMU frame)
[ERROR] Unhandled error in evo.main_rpe
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 94, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_rpe.py", line 311, in run
    args.t_offset,
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 232, in load_assoc_euroc_trajectories
    traj_ref = read_euroc_csv_trajectory(ref_file)
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 145, in read_euroc_csv_trajectory
    mat = np.array(csv_read_matrix(file_path, delim=",", comment_str="#")).astype(float)
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 61, in csv_read_matrix
    mat = [row for row in reader]
Error: new-line character seen in unquoted field - do you need to open the file in universal-newline mode?

[ERROR] evo module evo.main_rpe crashed - no logfile written (disabled)

  How can I solve it?
  Thank you.

MichaelGrupp, 2018-06-08:
Your data seems corrupted or not in the correct [format](https://github.com/MichaelGrupp/evo/wiki/Formats). Please check or upload it here.

xukuanHIT, 2018-06-09:
OK， Thank you.

closed 2018-06-09 by xukuanHIT