MichaelGrupp/evo #57: Kitti RPE  plot problem

goldenminerlmg, 2018-06-17:
Hi Mr...!
Thank you for your work! Here is a question when I use the following command to plot the rpe colortrajectory:
evo_rpe kitti KITTI_01_gt.txt KITTI_01_ndt.txt -va --plot --plot_mode xz --save_results result_01_RPE/ndt.zip
The origin point doesn't appear at the same pose with the ground truth. But i have tested other sequence on kitti and the result is correct. So what problem you think i have come across.
This is the result figure:
[image]
Thank you very much! Looking forward to your reply!

MichaelGrupp, 2018-06-17:
The `a` in `-va` is `--align`, which can lead to the different start coordinates. Other than that, there's a big drift and/or scale error that can have various reasons.

goldenminerlmg, 2018-06-18:
So it has no effect on the final result of "evo_res ......." , which is used to compare with other algorithms?

MichaelGrupp, 2018-06-18:
What do you mean by "it"?

If you're talking about the result of RPE with and without alignment: in the case of the relative pose error, global SE(3) alignment doesn't change the result because, as the name says, it only considers relative poses, not global ones (additionally correcting scale would change the translational RPE though).