MichaelGrupp/evo #58: Output plot images directly

hlzz, 2018-06-17:
Thanks for this great work, which I think is definitely a significant contribution to the community! My question is simple so bare with me if I am not using the template.

[image]

I can plot the above figures as the instruction shows. But sometimes I also want to save these trajectory and APE error figures directly without pausing or showing them. I am wondering if there is an option to do this? I checked the wiki careful and found this command

`evo_config set save_traj_in_zip true
`

but it seems that this is also not what I wanted because it generates the trajectories in for example .kitti suffix.

MichaelGrupp, 2018-06-18:
> Thanks for this great work, which I think is definitely a significant contribution to the community! My question is simple so bare with me if I am not using the template.

Thanks! Don't worry, the template is only needed for error or bug reports to provide information on how to reproduce errors. Simple questions don't need it ;)

---

You can use `--save_plot`, for example:
`evo_res < your files > --save_plot /path/to/plot.pdf`
This ouputs in pdf format by default, but you can change it with `evo_config set plot_export_format ...` to another format that is supported by matplotlib. There are also some examples of other settings you might want to tweak in the Wiki.

hlzz, 2018-06-21:
@MichaelGrupp I was able to run the example smoothly. But sometimes evo_traj gives me nothing, after I type

`evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz`

It just output an empty line without showing the plotting. But simply type `evo_traj kitti` gives me 

> usage: evo_traj kitti [-h] [-f] [-a] [-s] [--n_to_align N_TO_ALIGN]
>                       [--transform_left TRANSFORM_LEFT]
>                       [--transform_right TRANSFORM_RIGHT] [--invert_transform]
>                       [--ref REF] [--t_offset T_OFFSET]
>                       [--t_max_diff T_MAX_DIFF] [--merge] [-p]
>                       [--plot_mode {xy,yx,xz,zx,yz,xyz}]
>                       [--save_plot SAVE_PLOT]
>                       [--serialize_plot SERIALIZE_PLOT] [--save_as_tum]
>                       [--save_as_kitti] [--save_as_bag] [--no_warnings] [-v]
>                       [--silent] [--debug] [-c CONFIG]
>                       pose_files [pose_files ...]

In this case --silent -v -save_plot don't work either. I should have installed it right (I tried evo(1.0.2) build from source and from pypi on ubuntu 16.04 with python 2.7.12).

MichaelGrupp, 2018-06-21:
Just typing `evo_traj kitti` without any input file does nothing and shows only the usage message. That's fine.

Please give the full command that doesn't work for you.

hlzz, 2018-06-21:
@MichaelGrupp The commands in the README.md don't work for me, such as

`evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
`
`evo_ape kitti KITTI_00_gt.txt KITTI_00_ORB.txt -va --plot --plot_mode xz --save_results results/ORB.zip
`

They just print nothing but an empty line, but at least `evo` can show something.

> tianwei@tianwei:~/Documents/Projects/evo/test/data$ evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
> 
> tianwei@tianwei:~/Documents/Projects/evo/test/data$ mkdir results
> tianwei@tianwei:~/Documents/Projects/evo/test/data$ evo_ape kitti KITTI_00_gt.txt KITTI_00_ORB.txt -va --plot --plot_mode xz --save_results results/ORB.zip
> 
> tianwei@tianwei:~/Documents/Projects/evo/test/data$ evo_traj -h
> usage: evo_traj [-h] {kitti,tum,euroc,bag} ...
> 
> trajectory analysis and manipulation tool (c) michael.grupp@tum.de
> 
> positional arguments:
>   {kitti,tum,euroc,bag}
> 
> optional arguments:
>   -h, --help            show this help message and exit
> tianwei@tianwei:~/Documents/Projects/evo/test/data$ evo
> usage: evo [-h] {pkg,cat_log} ...
> 
> Python package for the evaluation of odometry and SLAM
> (c) michael.grupp@tum.de - license: run 'evo pkg --license'
> 
> Supported trajectory formats:
> * TUM trajectory files
> * KITTI pose files
> * ROS bagfile with geometry_msgs/PoseStamped topics
> * EuRoC MAV dataset groundtruth files
> 
> The following executables are available:
> 
> Metrics:
>    evo_ape - absolute pose error
>    evo_rpe - relative pose error
> 
> Tools:
>    evo_traj - tool for analyzing, plotting or exporting multiple trajectories
>    evo_res - tool for processing multiple result files from the metrics
>    evo_ipython - IPython shell with pre-loaded evo modules
>    evo_fig - (experimental) tool for re-opening serialized plots
>    evo_config - tool for global settings and config file manipulation
> 
>

MichaelGrupp, 2018-06-21:
Wow, that's strange. `evo` just uses the normal print function, whereas the "real" apps like `evo_traj` etc use Python's logging module. This will be hard to debug because I never had any problems on my PC, but let's  try...

* Is the command line output the only thing that doesn't work? (are the plot windows shown & output files written?)
* Run `evo_config set logfile_enabled`, run a few commands and check if it writes something to the logfile. If it does, only the console output is  broken. You can print the logfile with `evo cat_log` or alternatively open the file (path is shown by: `evo pkg --logfile`).

hlzz, 2018-06-21:
@MichaelGrupp 
1. No plot window is shown, nor output files written.
2. after `evo_config set logfile_enabled`,

> tianwei@tianwei:~/Documents/Projects/evo/test/data$ evo_config set logfile_enabled
> --------------------------------------------------------------------------------
> Old configuration:
> --------------------------------------------------------------------------------
> {
>     "logfile_enabled": false, 
>     "logging_format": "%(message)s", 
>     "plot_backend": "Qt4Agg", 
>     "plot_export_format": "pdf", 
>     "plot_figsize": [
>         6, 
>         6
>     ], 
>     "plot_fontfamily": "sans-serif", 
>     "plot_fontscale": 1.0, 
>     "plot_hideref": false, 
>     "plot_info_text": false, 
>     "plot_invert_xaxis": false, 
>     "plot_invert_yaxis": false, 
>     "plot_linewidth": 1.5, 
>     "plot_multi_cmap": "none", 
>     "plot_seaborn_palette": "deep", 
>     "plot_seaborn_style": "darkgrid", 
>     "plot_split": false, 
>     "plot_texsystem": "pdflatex", 
>     "plot_trajectory_cmap": "jet", 
>     "plot_usetex": true, 
>     "plot_xyz_realistic": true, 
>     "save_traj_in_zip": false, 
>     "table_export_data": "stats", 
>     "table_export_format": "csv", 
>     "table_export_transpose": true
> }
> 
> --------------------------------------------------------------------------------
> New configuration:
> --------------------------------------------------------------------------------
> {
>     "logfile_enabled": true, 
>     "logging_format": "%(message)s", 
>     "plot_backend": "Qt4Agg", 
>     "plot_export_format": "pdf", 
>     "plot_figsize": [
>         6, 
>         6
>     ], 
>     "plot_fontfamily": "sans-serif", 
>     "plot_fontscale": 1.0, 
>     "plot_hideref": false, 
>     "plot_info_text": false, 
>     "plot_invert_xaxis": false, 
>     "plot_invert_yaxis": false, 
>     "plot_linewidth": 1.5, 
>     "plot_multi_cmap": "none", 
>     "plot_seaborn_palette": "deep", 
>     "plot_seaborn_style": "darkgrid", 
>     "plot_split": false, 
>     "plot_texsystem": "pdflatex", 
>     "plot_trajectory_cmap": "jet", 
>     "plot_usetex": true, 
>     "plot_xyz_realistic": true, 
>     "save_traj_in_zip": false, 
>     "table_export_data": "stats", 
>     "table_export_format": "csv", 
>     "table_export_transpose": true
> }
> 

Then I run a command:

> tianwei@tianwei:~/Documents/Projects/evo/test/data$ evo_ape kitti KITTI_00_gt.txt KITTI_00_ORB.txt -va --plot --plot_mode xz --save_results results/ORB.zip
> 
> Open logfile? (y/n)
> y
> tianwei@tianwei:~/Documents/Projects/evo/test/data$ gvfs-open: /home/tianwei/.evo/evo.log: error opening location: Error when getting information for file '/home/tianwei/.evo/evo.log': No such file or directory
> 

It seems that the logfile doesn't exist...

hlzz, 2018-06-21:
@MichaelGrupp Thanks for your help. Fortunately I tried python3 and it works as expected. Note that python3-pyqt4 is needed for evo to work correctly, otherwise we may get `ImportError: No module named 'PyQt4'` error.

`sudo apt-get install python3-pyqt4 `

There may be other dependencies such as `python3-dev` and `libqt4-dev` but I am not sure. I will leave it for a future reference.

But I do remember that I can run evo with python 2.7 yesterday. Then all of a sudden it doesn't work, weird...

MichaelGrupp, 2018-06-21:
Cool that you could find a solution. I was about to ask about your system and dependencies, I still don't know what could have happened. 

Regarding Qt: normally, it checks during installation if it is installed, and if it is not installed it uses Tk which should work with standard Python installations.

closed 2018-06-22 by MichaelGrupp

deepmeng, 2018-12-07:
OK, it was a hard process for me. The problems and solutions are shotly summarized as following.
1. When anyone meets the error of empty output after inputting evo_* command, you can try to install python*-pyqt*. 
2.Then, when anyone meets the error of “from matplotlib.backends.backend_tkagg import FigureCanvasTkAgg, NavigationToolbar2Tk ImportError: cannot import name 'NavigationToolbar2Tk”, please upgrade the matplotlib in time.

Finally, I would like to thank the author for this powerful tool. Thanks again.