MichaelGrupp/evo #59: evo_traj tum issue

amranov, 2018-06-18:
Hi , 
i am trying to plot KeyFrameNavStateTrajectory.txt  usuing evo_traj tum KeyFrameNavStateTrajectory.txt but i get this error 
amrani@amrani:~/testResults$ evo_traj tum KeyFrameNavStateTrajectory.txt 
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 94, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_traj.py", line 177, in run
    trajectories.append((traj_file, file_interface.read_tum_trajectory_file(traj_file)))
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 71, in read_tum_trajectory_file
    mat = np.array(csv_read_matrix(file_path, delim=" ", comment_str="#")).astype(float)
ValueError: could not convert string to float: 

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)
any suggession how to fix this errors ?

MichaelGrupp, 2018-06-18:
Looks like a classic "data in the wrong format" issue as this one for example: https://github.com/MichaelGrupp/evo/issues/55

If you need more help, please upload a reproducable example using [issue template](https://github.com/MichaelGrupp/evo/blob/master/ISSUE_TEMPLATE.md)

amranov, 2018-06-19:
Dear Michael , evo_traj work properly when i plot FrameTrajectory_KITTI_Format.txt (ORB_SLAM) output , but here i am trying to plot LearVIORB_SLAM TUM format . 
this is error i get after conversting the format into Kitti 
root@amrani:~/testResults# evo_traj tum KeyFrameNavStateTrajectory.txt --save_as_kitti 
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/evo/entry_points.py", line 94, in launch
    main_module.run(args)
  File "/usr/local/lib/python2.7/dist-packages/evo/main_traj.py", line 177, in run
    trajectories.append((traj_file, file_interface.read_tum_trajectory_file(traj_file)))
  File "/usr/local/lib/python2.7/dist-packages/evo/tools/file_interface.py", line 71, in read_tum_trajectory_file
    mat = np.array(csv_read_matrix(file_path, delim=" ", comment_str="#")).astype(float)
ValueError: could not convert string to float: 

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

MichaelGrupp, 2018-06-19:
> but here i am trying to plot LearVIORB_SLAM TUM format

Whatever that is - it's certainly not the official TUM benchmark format that evo supports.

> error i get after conversting the format into Kitti

Your input format for the conversion is still not the correct TUM format, how should this work?

---

Upload something to reproduce, or nobody can help you.

amranov, 2018-06-19:
Dear michael , sorry for bothering im quite new in this field here my trajectory.txt file 
[KeyFrameNavStateTrajectory.txt](https://github.com/MichaelGrupp/evo/files/2114551/KeyFrameNavStateTrajectory.txt)

MichaelGrupp, 2018-06-19:
The file has 2 problems:
* trailing space at the end of each line - space is the delimiter of the TUM format and a delimiter without data after it leads to an error
* too many columns, looks like there is some additional data that we don't expect

If you you have Linux (probably also works on Mac) you can use `cut` to create a clean copy with only the first 8 columns:
```bash
cut --delimiter " " --fields 1-8 KeyFrameNavStateTrajectory.txt > clean.tum
```

This file is then valid according to `evo_traj tum clean.tum --full_check `.

amranov, 2018-06-20:
THank you so much 
it works properly now

closed 2018-06-20 by amranov

jingshaojing, 2021-01-21:
I have try this  method, but it doesnot work