MichaelGrupp/evo #62: How to get rotation error?

WongRZ, 2018-06-25:
Dear author,
Through command "evo_ape euroc", i can get the translation error of my own trajectory with ground-truth. But what about rotation error, how can i get it. Can you give me a hint?
Thanks a lot.

MichaelGrupp, 2018-06-25:
Hi,

this can be done with the `--pose_relation` argument (default: `translation_part`). There are three pose relation arguments for the rotation error:

* `angle_rad`: angle in radians
* `angle_deg`: angle in degrees
* `rot_part`: unitless, uses a matrix norm for the rotation matrix part of the pose

Formulas for this are documented in the Jupyter notebook: https://github.com/MichaelGrupp/evo/blob/master/notebooks/metrics.py_API_Documentation.ipynb

An interesting mathematical property of these rotation metrics is that the values only differ by scale :)

---

In my opinion it makes usually more sense to measure the relative angular error than the absolute one, for example the angular drift per meter: `evo_rpe tum <ref> <estimate> --pose_relation angle_deg --delta 1.0 --delta_unit m`

WongRZ, 2018-06-25:
Thanks a lot, dear author. Another question is, the difference between absolute pose error and relative pose error in this app. And why do you think it makes usually more sense to measure the relative angular error than the absolute one. 
Thanks again.

MichaelGrupp, 2018-06-25:
Take a look at the documentation (Wiki, Jupyter notebooks). The concept of absolute and relative error is also covered in papers like:
```
J. Sturm, N. Engelhard, F. Endres, W. Burgard, D. Cremers: 
A Benchmark for the Evaluation of RGB-D SLAM Systems 
In Proc. of the International Conference on Intelligent Robot Systems (IROS), 2012.
```

WongRZ, 2018-06-26:
Thanks. I used "evo_ape euroc <ref> <estimate> -r rot_part" command to calculate the rotate errors. But the results are unexpected. Neither the result of ORB_SLAM2 nor my own result was  right. The RMSE of them are more than 70 degrees. I think the format of my file may be wrong. But I'm just following the TUM format, the same as the ORB_SLAM's. Attached is the result of ORB_SLAM, would you please try it?

MichaelGrupp, 2018-06-26:
This is probably related to https://github.com/MichaelGrupp/evo/issues/23

The different coordinate frames of groundtruth and SLAM (IMU, camera) are problematic if you do absolute rotation comparisons without properly transforming everything beforehand. One of the reasons why relative error is better in that case.

closed 2018-06-30 by MichaelGrupp

chenwinki, 2018-07-04:
@MichaelGrupp  Dear author : I use "evo_rpe tum data.tum clean.tum -a --delta 1.0 --delta_unit m --pose_relation angle_deg -p" to computer angular drift per meter ,but the result id not unexpected:
[image]
[image]
and the estimate file : 
[clean.txt](https://github.com/MichaelGrupp/evo/files/2161808/clean.txt)
reference file : 
[data.txt](https://github.com/MichaelGrupp/evo/files/2161810/data.txt)
Could you give me some hint ? thank you very much!