MichaelGrupp/evo #63: RMSE

einspyon, 2018-06-25:
Hi, really thanks for your all  nice and generous work!
But, there's a small problem which troubles me, looking for your help.

When I run evo_rpe with delta_unit -- 'f', I got a small RMSE error, such as:
[image]
[image]

But when I run evo_rpe with delta_uint -- 'm', it returns s really big RMSE value, eg:
[image]
[image]

So, In general, such as in most of papers, How do I choose the delta_unit for a suitable comparison ?
Thanks!

MichaelGrupp, 2018-06-25:
Of course the results are different:

* `--delta 1 --delta_unit f` is comparing the relative poses between each (consecutive) pose pair that has a distance of 1 "frame"
* `--delta 1 --delta_unit m` is comparing the relative poses between each (consecutive) pose pair that has a distance of 1 meter

---

One other remark: `--pose_relation full` is a unitless matrix norm, I wouldn't use this in a paper (rather separate angle and translation).

einspyon, 2018-06-26:
Hi, thanks for your patience @MichaelGrupp !  
I eager to know which "delta_unit" should I choose for fair comparison, for example: 
[image]
I want to compare my results with the above paper results,  but when  "--delta 1 --delta_unit m" is in choice,  my rmse is about 1.28, it's so large.   When " --delta 1 --delta_unit f " is in choice, my rmse is about 0.034, a little confused.
Thanks anyaway!!!

MichaelGrupp, 2018-06-26:
Well... what does the paper use? How you configure the metrics is compeletely up to you.

einspyon, 2018-06-27:
I don't know paper's criteria, it's not mentioned, but I think a consistent delta unit in my comparison would be fair, thanks @MichaelGrupp

closed 2018-06-27 by einspyon

MichaelGrupp, 2018-06-27:
Then use a delta based on spatial distance, i.e. meters or degrees. "Frames" is not consistent, it highly depends on how far frames are apart from each other.

By the way: consider `--all_pairs`, it makes it statistically more robust.