MichaelGrupp/evo #66: When using evo_traj, the attitude Angle cannot be aligned.

mengqingyu123, 2018-06-27:
Bug / Error report

Description:Hello, I use https://github.com/jingpang/LearnVIORB test euroc data set, the output results are no problem, but I found that when using attitude Angle contrast with evo_traj not fitting.

What's wrong with my operation? Thank you!

mengqingyu123, 2018-06-27:
[image]
[image]
[image]

This is the result of running.

MichaelGrupp, 2018-06-27:
Looks like a duplicate of: https://github.com/MichaelGrupp/evo/issues/23

mengqingyu123, 2018-06-28:
I see. Thank you for your help.

closed 2018-06-28 by mengqingyu123

mengqingyu123, 2018-07-03:
@chenwinki I consulted MichaelGrupp suggestion, absolute euler Angle error is not easy to understand and have limitations, so I use the evo_rpe get some indicators of euler Angle error, rather than directly compare the pitching Angle, course Angle and roll Angle.

chenwinki, 2018-07-03:
@mengqingyu123 thank you~~~

MichaelGrupp, 2018-07-03:
> I use the evo_rpe get some indicators of euler Angle error

This doesn't use Euler angles. Check the formulas in `notebooks/metrics.py_API_Documentation.ipynb`.

mengqingyu123, 2018-07-03:
@MichaelGrupp Uh... Sorry, I want to ask whether the command "evo_rpe tum MH05_groundtruth. TXT msckf_mh05.txt-a --pose_relation angle_deg-p" outputs some indicators of euler Angle. Why did you say you didn't use euler's Angle? I'm sorry to bother you with this question.

MichaelGrupp, 2018-07-03:
> Why did you say you didn't use euler's Angle?

Because it's not using them. Euler angles are only used for easy-to-interpret visualization in the plot of `evo_traj`. All angular metrics (`--pose_relation angle_deg / angle_rad`) calculate the rotation angle using the logarithmic map of the SO(3) rotation matrix, as shown in the formulas mentioned above. Or check what the code does in https://github.com/MichaelGrupp/evo/blob/master/evo/core/metrics.py , e.g.:

```python
...
elif self.pose_relation == PoseRelation.rotation_angle_rad:
    self.error = np.array([abs(lie.so3_log(E_i[:3, :3])) for E_i in self.E])
...
```