MichaelGrupp/evo #70: the problem of format of file.

chenwinki, 2018-07-03:
thank you for your contribution！
I have two file , (1)data.tum is transformed from **groundtruth** data.csv in EuRoC (2) msckf_traj.txt is the output of my system contain 8 elements(timestamp x y z q_x q_y q_z q_w)
I run "evo_traj tum msckf_traj.txt" , but error like follow:
[image]
how can i transform msckf_traj.txt to tum format and compare with data.tum? looking forward to your replay!

MichaelGrupp, 2018-07-03:
Why don't you just upload your files?

chenwinki, 2018-07-03:
sorry , this is msckf_traj.txt , 
[msckf_traj.txt](https://github.com/MichaelGrupp/evo/files/2157903/msckf_traj.txt)

 the output of my system contain 8 elements(timestamp x y z q_x q_y q_z q_w)

MichaelGrupp, 2018-07-03:
The file has trailing whitespace in each line (a space too much). See [here](https://github.com/MichaelGrupp/evo/issues/59#issuecomment-398339844) how you could clean your data:

```bash
cut --delimiter " " --fields 1-8 msckf_traj.txt > clean.tum
```

MichaelGrupp, 2018-07-03:
But there are more problems, the poses are not correctly sorted by time. At least one jump back in time is reported by:

```
evo_traj tum clean.tum --full_check 
--------------------------------------------------------------------------------
name:	clean
infos:
	duration (s)	0.0
	nr. of poses	2893
	path length (m)	60.1378313639
	pos_end (m)	[0.510796  0.244681  0.0302371]
	pos_start (m)	[ 0.596255    0.217576   -0.00800546]
	t_end (s)	1403715418.812143
	t_start (s)	1403715418.812143
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	wrong, not ascending or duplicates
stats:
	error - bad timestamps: 1403715418.812143 & 1403715274.262143
```

chenwinki, 2018-07-03:
thank you very much!

chenwinki, 2018-07-03:
Dear author ，I "evo_traj tum clean.tum --ref data.tum -p" compare the two files ,the result is follow , the result seems similar but  opposite, can you direct me a way? thank you !
[image]

closed 2018-07-04 by MichaelGrupp