MichaelGrupp/evo #73: Allignment proplem and question with coordinate

imrasp, 2018-07-06:
I am working on gathering ground-truth from GPS and IMU, I have got problem of alignment mismatch as in the image below resulting from `evo_ape tum interpolate_ned_output_linear_cut.txt groundtruth2.txt --plot -va` command
[image]

I follow TUM format ( https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats) except the coordinate, which I use NED coordinate and I think that Umeyama's method can handle this.
My current format follow TUM is `timestamp tx ty tz qx qy qz qw` where
* timestamp (float) gives the number of seconds since the Unix epoch.
* tx ty tz (3 floats) give the position of sensors in NED coordinate in meter.
* qx qy qz qw (4 floats) give the orientation of the sensors in quaternion components.

Do I have to convert to coordinate the camera with respect to the world origin as defined by the motion capture system ( Is this mean camera coordinate where origin at center of the camera and x-right, y-up, z-forward).

Could you please suggest me on how to fix this

Full-check result are:
```
evo_traj tum groundtruth2.txt --full_check --plot
--------------------------------------------------------------------------------
name:	groundtruth2
infos:
	duration (s)	430.330169916
	nr. of poses	13562
	path length (m)	1035.2306277
	pos_end (m)	[ 31.49409783 -89.38959442  -1.95580373]
	pos_start (m)	[ 0.  0. -0.]
	t_end (s)	1526796495.99
	t_start (s)	1526796065.66
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	ok
stats:
	v_avg (km/h)	8.674677
	v_avg (m/s)	2.409632
	v_max (km/h)	27.486197
	v_max (m/s)	7.635055
	v_min (km/h)	0.283611
	v_min (m/s)	0.078781
```
and 
```
evo_traj tum interpolate_ned_output_linear_cut.txt --full_check --plot
--------------------------------------------------------------------------------
name:	interpolate_ned_output_linear_cut
infos:
	duration (s)	480.318549871
	nr. of poses	15142
	path length (m)	1164.95321947
	pos_end (m)	[ 0.65590576  0.39332212 -2.68988697]
	pos_start (m)	[ 0.          0.         -0.69342154]
	t_end (s)	1526796496.05
	t_start (s)	1526796015.73
checks:
	SE(3) conform	yes
	array shapes	ok
	nr. of stamps	ok
	quaternions	ok
	timestamps	ok
stats:
	v_avg (km/h)	8.740063
	v_avg (m/s)	2.427795
	v_max (km/h)	27.999939
	v_max (m/s)	7.777761
	v_min (km/h)	0.000119
	v_min (m/s)	0.000033

```

**Input files:**
[groundtruth2.txt](https://github.com/MichaelGrupp/evo/files/2168680/groundtruth2.txt) - position from GPS converted to NED
[interpolate_ned_output_linear_cut.txt](https://github.com/MichaelGrupp/evo/files/2168682/interpolate_ned_output_linear_cut.txt)

**Additional information:**
* `evo pkg --version`: v0.9.9
* Python version : 2.7.12 
* Ubuntu 16.04 
* output of `evo_config show --brief --no_color`: 
```
{
    "logfile_enabled": false, 
    "logging_format": "%(message)s", 
    "plot_backend": "TkAgg", 
    "plot_export_format": "pdf", 
    "plot_figsize": [
        6, 
        6
    ], 
    "plot_fontfamily": "sans-serif", 
    "plot_fontscale": 1.0, 
    "plot_hideref": false, 
    "plot_info_text": false, 
    "plot_invert_xaxis": false, 
    "plot_invert_yaxis": false, 
    "plot_linewidth": 1.5, 
    "plot_multi_cmap": "none", 
    "plot_seaborn_palette": "deep", 
    "plot_seaborn_style": "darkgrid", 
    "plot_split": false, 
    "plot_texsystem": "pdflatex", 
    "plot_trajectory_cmap": "jet", 
    "plot_usetex": false, 
    "plot_xyz_realistic": true, 
    "save_traj_in_zip": false, 
    "table_export_data": "stats", 
    "table_export_format": "csv", 
    "table_export_transpose": true
}

```

MichaelGrupp, 2018-07-06:
Wow that sounds like a cool setup.

Your file format look good to me, I think it's fine to use the NED coordinates. Using Umeyama alignment also sounds ok.

I don't know how you recorded the data, but could it be that you have a time synchronization issue? Time synchronization is very critical for accurate ground truth collection, ideal would be a common clock. It could be that there's a time offset that has the effect that the trajectory poses are not matched correctly before Umeyama alignment.

Just to illustrate, I tried a guessed value for the time offset and it looks like the trajectory matching makes more sense. Of course this value is just a wild guess.
```bash
evo_traj tum groundtruth2.txt interpolate_ned_output_linear_cut.txt --ref groundtruth2.txt -ap --t_offset -21 --plot_mode xy
```
[image]

MichaelGrupp, 2018-07-06:
If you run into problems with using the offset parameters, it's probably related to issue #74, which I just noticed while playing with your data.

imrasp, 2018-07-08:
Thank you so much. your answer put me through to the right track to find root of the problem. I recheck my data again and I found that problem is not result from timestamp but might cause from mistake on conversion and interpolation processes now result from `evo_ape tum interpolate_ned_output_linear_cut.txt groundtruth2.txt --plot -va` is 

[image]

closed 2018-07-08 by imrasp