MichaelGrupp/evo #77: How do you do it?

xukuanHIT, 2018-07-19:
Hi, thank you for your amazing work, I want to know how it works now, but the python code is little difficult to me , Can give me some paper or document about evo?

MichaelGrupp, 2018-07-19:
There's no dedicated paper about this software. Only the documentation here and the code, which is commented in most places. If there's something missing in the documentation, please ask a bit more specific what you are interested in.

MichaelGrupp, 2018-07-20:
It aligns the "mono" trajectory to the reference trajectory by computing the optimal similarity transformation (rotation, translation, scale) that minimizes the distance between the points of both trajectories. The method that's used is Umeyama's method, which can compute this transformation in closed form. This requires the points to be registered to each other, but this is the case here because we register the trajectory points by matching their timestamps.

See [here](https://github.com/MichaelGrupp/evo/wiki/Metrics#alignment) for how to use it and [here](https://github.com/MichaelGrupp/evo/blob/master/evo/core/geometry.py#L28) for the generic implementation of this method. The title of the paper of the method is in the comment there.

xukuanHIT, 2018-07-20:
OK, thank you very much

closed 2018-07-20 by xukuanHIT

bryantaoli, 2020-03-15:
But the scale in mono SLAM is chaning every time, but you only estimate one scale, does it make some errors?

MichaelGrupp, 2020-03-16:
The scale correction in evo is just a global "best fit" over the whole trajectory to allow comparison with groundtruth. Avoiding scale drift is the job of your SLAM system.