MichaelGrupp/evo #79: Live evo plot with orbslam.

rahulsharma11, 2018-08-07:
Hi,
I am running orbslam2 on my jetson tx2. I am currently sending live 3d pose and coordinate data through jetson to my laptop over socket. I am able to plot 2d coordinate data but not able to do that with 3d.

Currently evo_traj is plots the data by reading the generated text file. Is it possible to have that realtime?

Thanks.

MichaelGrupp, 2018-08-07:
> Currently evo_traj is plots the data by reading the generated text file. Is it possible to have that realtime?

Not out of the box and it would require you to implement a custom solution yourself.

You just want to plot the ORB SLAM trajectory? Then I don't see why you should use evo for that, there are way better solutions for that. E.g. RViz if you are using ROS or Pangolin (the standard visualization of ORB SLAM).

rahulsharma11, 2018-08-07:
Hi, I am not using ROS/Pangoline . I just wanted to plot the raw 3d data that i am getting from orbslam. Instead i was using matplotlib for that. 
Anyway i just wanted to know if i can do that with evo_traj or not.
Thanks for the confirmation.

closed 2018-08-20 by MichaelGrupp