MichaelGrupp/evo #80: wrong format？

jiazhibo, 2018-08-14:
**Description:**

**Command:**
```
# evo_traj tum result.tum -p --plot_mode=xy
```

**Console output:**
```

# [ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/jiazhibo/.local/lib/python2.7/site-packages/evo/entry_points.py", line 87, in launch
    main_module.run(args)
  File "/home/jiazhibo/.local/lib/python2.7/site-packages/evo/main_traj.py", line 177, in run
    trajectories.append((traj_file, file_interface.read_tum_trajectory_file(traj_file)))
  File "/home/jiazhibo/.local/lib/python2.7/site-packages/evo/tools/file_interface.py", line 76, in read_tum_trajectory_file
    file_path, delim=" ", comment_str="#")).astype(float)
ValueError: could not convert string to float: 

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

```

**Additional files:**
[result.txt](https://github.com/MichaelGrupp/evo/files/2285652/result.txt)
[groundtruthSync.txt](https://github.com/MichaelGrupp/evo/files/2285668/groundtruthSync.txt)

Please give also the following information:
I think there are some problems with the data format I output, but I don't know how to solve it. When I convert to a tum file, I will lose a lot of data.
groundtruthSync.txt is the standard data format.

MichaelGrupp, 2018-08-17:
If you use a custom format, you need to convert it to one of the supported formats - see the documentation. If you'd "lose data", just make a copy?

MichaelGrupp, 2018-08-17:
Regarding your error: check if there's trailing whitespace. There's also another issue with a similar error here on GitHub.

jiazhibo, 2018-08-20:
Thank you very much.
I found that when there is extra space between my two data, the last data of that line will be deleted.
When there is only one space between each data, the error is gone.

closed 2018-08-26 by MichaelGrupp