MichaelGrupp/evo #85: How to adjust the parameter of the simulation map

Junkai-Liu, 2018-09-03:
How to turn the background of the simulation map into pure white?Otherwise, how to adjust legend of the map, such as line width、color and so on.

MichaelGrupp, 2018-09-03:
See this other issue, it had a similar question: https://github.com/MichaelGrupp/evo/issues/14

How to show and change the available parameters:
https://github.com/MichaelGrupp/evo/wiki/Configuration

A dummy example for changing plot parameters:
https://github.com/MichaelGrupp/evo/wiki/Plotting

Have a look at the parameters that you can change with `evo_config show` and play around with `evo_config set`. Of course there are things you can't change this way, so you might need to do custom coding or request a new parameter that you would like to have implemented. Depends on what you want.

MichaelGrupp, 2018-09-05:
Closing as answered, feel free to re-open if you have more questions.

closed 2018-09-05 by MichaelGrupp