MichaelGrupp/evo #87: pose segment meaning

hlzz, 2018-09-05:
[image]
I've been using evo recently for my research and I love it! Thanks again for your great tool! May I ask the meaning of line segments in pose trajectory, as you can see from the picture. I know that the color represents the severity of the error, but what about the length of line segment for each frame? Thanks.

MichaelGrupp, 2018-09-05:
> I've been using evo recently for my research and I love it! Thanks again for your great tool!

Thanks! :smile: 

> what about the length of line segment for each frame?

I'm not sure if I understand your question correctly, but a line segment just connects two x,y,z positions of the trajectory. Thus, the length is only the distance between two poses of the trajectory.

hlzz, 2018-09-05:
I mean the trajectory should be a single curved line, like the dotted reference trajectory. But actually when you zoom in, the trajectory is composed of a sequence of zig-zag line segments. I wonder what does this represent? Can I draw the trajectory like the reference one? (only one curved line).

MichaelGrupp, 2018-09-05:
Hmm... this is your data, if the trajectory has a zig-zag pattern you will see it also here.

You should also see this pattern when you plot the trajectory with `evo_traj`, do you? If you want you can also upload your files here.

hlzz, 2018-09-12:
[09_full.txt](https://github.com/MichaelGrupp/evo/files/2375851/09_full.txt)
[10_full.txt](https://github.com/MichaelGrupp/evo/files/2375852/10_full.txt)
I see... The pose data is generated by concatenating relative poses. Maybe I should do a motion averaging after concatenating these pose snippets.

hlzz, 2018-09-12:
Anyway, I should do motion averaging to remove the zig-zag patterns of the poses. Thanks a lot.

closed 2018-09-12 by hlzz