MichaelGrupp/evo #100: fail to export euroc format as other formats

rhklite, 2018-10-13:
I followed the steps in [wiki](https://github.com/MichaelGrupp/evo/wiki/Formats#saving--exporting-to-other-formats), however the save is unsuccessful. Please see screenshot below,

[image]

Although i was able to convert between other formats, such as tum->ros, tum->kitti etc. 

I did `pip install --upgrade numpy matplotlib scipy pandas seaborn natsort argcomplete colorama pygments enum34 --user` to ensure all the packages are up to date.

MichaelGrupp, 2018-10-13:
I could reproduce this, it's a bug. I accidentally removed an `else` in the last commit :wink: 

Thanks for reporting this.

closed 2018-10-13 by MichaelGrupp

rhklite, 2018-10-14:
Thank you! It worked. Your program is really amazing!

commit 24069a08 (2018-10-13): Correct typo to fix #100. (#101)