MichaelGrupp/evo #105: MakeUmeyama method results accessible

TBondat, 2018-10-23:
Thanks for the very useful tool!
I maybe missed it but i didn't see a way to export the results of Umeyama process (in trajectory.py, these are r_a, t_a and s). They are displayed, but are not accessible or stored in trajectory data returned.

Is it possible to make the data accessible?

MichaelGrupp, 2018-10-23:
> They are displayed, but are not accessible or stored in trajectory data returned.

Could you explain what you mean with accessible in your case? For example, would it be enough if the [trajectory alignment function](https://github.com/MichaelGrupp/evo/blob/9d78b2fccaeaf6ef82d8d9683e5c8ea007ad8324/evo/core/trajectory.py#L366) returned these values? Storing the transformations as metadata in a trajectory object would be also possible, there is a [`meta` dictionary](https://github.com/MichaelGrupp/evo/blob/9d78b2fccaeaf6ef82d8d9683e5c8ea007ad8324/evo/core/trajectory.py#L50) that can be used for additional data (I use it to keep frame IDs of ROS data for example).

If you want that data to be saved in the default command line tools, this would be a bit more tricky.

TBondat, 2018-10-23:
yes, it is exactly that: if align_trajectory could send back the data, like adding them in traj_aligned, 
then in our file calling `traj_est_aligned_scaled = trajectory.align_trajectory(traj_est, traj_ref,correct_scale=True)`  for instance, one can reuse the umeyama data.
Like using `traj_est_aligned_scaled.r_a`.

The idea is to have it accessible in the user master function calling trajectory.align_trajectory()!

MichaelGrupp, 2018-10-23:
Then I would go for returning the values from the function. Since I don't want to break the API, I would add a new optional function parameter `return_parameters=False` [here](https://github.com/MichaelGrupp/evo/blob/9d78b2fccaeaf6ef82d8d9683e5c8ea007ad8324/evo/core/trajectory.py#L366) and if it's True, the function returns `return traj_aligned, r_a, t_a, s`. The default should still return only `traj_aligned`.

Then you can still store the values in the `meta` dict of the trajectory if you want, but it's not the default.

What do you think? If you want, you can test something like this and create a pull request.

TBondat, 2018-10-23:
yes, optional feature would be ok. Apparently, this is not a common need to get these values, so i suppose majority of people don't need it!

Regarding Github, i'm not familiar with how works all the update process, sorry.

TBondat, 2018-10-23:
thanks for the link, i will look at that!

MichaelGrupp, 2018-10-29:
I added it to the latest version.

closed 2018-10-29 by MichaelGrupp

TBondat, 2018-10-29:
oh ok, thanks! i will test it this week.

PR #108 (2018-10-29): Add return_parameters to align_trajectory()
Closes #105
commit 0fb8baa6 (2018-10-29): Add return_parameters to align_trajectories() (#108)

Closes #105