MichaelGrupp/evo #109: how to evaluate RPE rotation part on kitti?

goldenminerlmg, 2018-11-03:
Hi authors,
Thanks for your nice job! I have a question now that I want to compare my result on kitti. The default parameter used seems to RPE w.r.t. translation part (m):
evo_rpe kitti KITTI_00_gt.txt KITTI_00_test.txt -va --plot --plot_mode xz --save_results result_00_RPE/test.zip>

I saw other issues  #67 and #62 , that seems to use this command: 
evo_rpe tum reference.txt estimate.txt --pose_relation angle_deg --delta 1 --delta_unit m

but when I change my command as:
evo_rpe kitti KITTI_00_gt.txt KITTI_00_test.txt --pose_relation angle_deg --delta 1 --delta_unit m

there are errors:
Traceback (most recent call last):
  File "/home/xx/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/entry_points.py", line 89, in launch
    main_module.run(args)
  File "/home/xx/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/main_rpe.py", line 245, in run
    est_name=est_name,
  File "/home/xx/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/main_rpe.py", line 176, in rpe
    rpe_metric.process_data(data)
  File "/home/xx/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/core/metrics.py", line 296, in process_data
    abs(lie.so3_log(E_i[:3, :3])) * 180 / np.pi for E_i in self.E
  File "/home/xx/.virtualenvs/evaluation/local/lib/python2.7/site-packages/evo/core/lie_algebra.py", line 73, in so3_log
    raise LieAlgebraException("matrix is not a valid SO(3) group element")
LieAlgebraException: matrix is not a valid SO(3) group element

Is this because evo_rpe  --pose_relation only suit for tum-like data, and kitti's "matrix is not a valid SO(3) group element"? And how can I evaluate rotation of kitti by this tool.
Thank you very much!

MichaelGrupp, 2018-11-03:
Yes, you need to change `--pose_relation` to do that. This works for all input formats, also for KITTI.

What happens here:

* the angular error between the relative pose pairs is calculated using the rotation matrices (SO(3) matrices)
* for each rotation matrix, we check if it is a valid SO(3) matrix before calculating the angle. This happens here: https://github.com/MichaelGrupp/evo/blob/0fb8baa6021239617d816e42695d2202850c83ca/evo/core/lie_algebra.py#L72

In your case, it throws an error at this check. Can you check your input data? For the KITTI format, the rotation matrices are directly read from the input file - if there's an invalid one in your file it will lead to this error.

You can see if a file is okay with:
```bash
evo_traj kitti <your file> --full_check
```

hahakid, 2021-02-01:
@MichaelGrupp sorry for bother you, I used the last version of evo.
I import the source file in my code, and face the same problem:  not a valid SO(3) only when using **rotation_angle_rad/degree**
I use evo_traj kitti <your file> --full_check to check my pose files and get normal result (I think they are ok)

evo_traj kitti ../../results/test/00_d.txt --full_check
--------------------------------------------------------------------------------
name:   00_d
infos:
        nr. of poses    4540
        path length (m) 3736.4001953688794
        pos_end (m)     [-6.031866   0.1486681 94.56277  ]
        pos_start (m)   [ 0.000000e+00 -1.455192e-11  0.000000e+00]
checks:
        SE(3) conform   yes
        array shapes    ok
        quaternions     ok
stats:
-------------------------------------------------------------------------------
Moreover, I add the times to KITTI data and form PoseTrajectory3D structure (as follows), they works find with other PoseRelation.
[image]

MichaelGrupp, 2021-02-01:
@hahakid Please provide a reproducible example - i.e.: input files & command that was used.

hahakid, 2021-02-02:
@MichaelGrupp as I further test, when I use the kitti GT (currently, 00/poses.txt ), it will cause this problem. My prodict pose does not cause this problem.