MichaelGrupp/evo #124: AttributeError: 'AnsiCodes' object has no attribute 'LIGHTRED_EX'

slowlythinking, 2019-01-14:
When I run evo_config, I got a problem (as shown below). And when I run the demo in source code, such as "evo_ape kitti ../data/KITTI_00_gt.txt ../data/KITTI_00_ORB.txt" or "evo_traj tum ../data/fr2_desk_ORB.txt", I got nothing, not results, not errors. I didn't get any error message in my installation process of evo.
I have tried reinstallation, still can not fix it.
thx for help.

**Command:**
```
evo_config -h
```
**Console output:**
```
Traceback (most recent call last):
  File "/EvaluateTools/venv/bin/evo_config", line 7, in <module>
    from evo.main_config import main
  File "/EvaluateTools/venv/local/lib/python2.7/site-packages/evo/main_config.py", line 37, in <module>
    from evo.tools import log, user, settings
  File "/EvaluateTools/venv/local/lib/python2.7/site-packages/evo/tools/log.py", line 32, in <module>
    Fore.LIGHTRED_EX, Fore.RESET)
AttributeError: 'AnsiCodes' object has no attribute 'LIGHTRED_EX'
```
**Additional files:**
evo 1.3.0
Python2.7.6(used virtualenv)
Ubuntu14.04
Don't change the source code.
```

MichaelGrupp, 2019-01-14:
Could be the same problem as here: https://github.com/kivy/buildozer/issues/366

I would try this in the virtual environment:
```
pip install --upgrade colorama
```

Maybe Ubuntu 14.04 has an older version of colorama already installed with `apt` and it doesn't upgrade it in your virtual environment when you use `pip` without `--upgrade`.

slowlythinking, 2019-01-14:
> Could be the same problem as here: [kivy/buildozer#366](https://github.com/kivy/buildozer/issues/366)
> 
> I would try this in the virtual environment:
> 
> ```
> pip install --upgrade colorama
> ```
> Maybe Ubuntu 14.04 has an older version of colorama already installed with `apt` and it doesn't upgrade it in your virtual environment when you use `pip` without `--upgrade`.

Thanks for reply. I'll try your approach tomorrow and let you know the result.

slowlythinking, 2019-01-15:
It worked for me. Thanks very much.
But there is another error:
When I install evo from PyPi, everything is fine. But when I install it from source, evo can not plot the result (as shown below).
**Command:**
```
evo_ape tum groundtruth.txt KeyFramePose.txt -va --plot --plot_mode xz --save_results result/SPTAM.zip
```
**Console output:**
```
Synchronizing trajectories...
Found 920 of max. 3156 possible matching timestamps between...
	groundtruth.txt
and:	KeyFramePose.txt
..with max. time diff.: 0.01 (s) and time offset: 0.0 (s).
--------------------------------------------------------------------------------
Aligning using Umeyama's method...
Rotation of alignment:
[[-0.68356877  0.22501349 -0.69433613]
 [ 0.72958001  0.23819411 -0.64107455]
 [ 0.02113636 -0.9447923  -0.3269874 ]]
Translation of alignment:
[-1.07970171  1.07780341  1.10510216]
Scale correction: 1.0
--------------------------------------------------------------------------------
Compared 920 absolute pose pairs.
Calculating APE for translation part pose relation...
--------------------------------------------------------------------------------
APE w.r.t. translation part (m)
(with SE(3) Umeyama alignment)

       max	1.278829
      mean	0.717010
    median	0.703018
       min	0.190018
      rmse	0.791122
       sse	575.804423
       std	0.334322

--------------------------------------------------------------------------------
Plotting results... 
[ERROR] unknown settings parameter: plot_reference_linestyle
```

MichaelGrupp, 2019-01-15:
Or use the new `v1.3.1` which is now released.

closed 2019-01-15 by MichaelGrupp