MichaelGrupp/evo #128: Actually I'm confused it's a bug or just my mistake,cause there's no plot,no printed information and no error

ghost069, 2019-01-17:
**Description:**
Well,basically it's what I said in the title,it shows nothing after I run the test command,every test command.I mean,that's not normal,right?

**Command:**
I installed evo from the source,python version 2.7.14

```
pip install . --upgrade --no-binary evo

```
Then I entered test/data,run

```
evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
and all the other examples,and it simply end like that
```

**Console output:**
[image]

Also,there is no saved file found in results/,basically nothing happened after this commands.But I can still import evo in python and run evo_ape euroc -h,so it's really confusing to me what's going on

MichaelGrupp, 2019-01-17:
In #99 someone described a similar issue. I still don't know if colorama was really causing this, but he said he could fix it by upgrading colorama.

If you pull the latest code, a minimum required colorama version is required and should be installed. Don't know if this helps, but you can try.

ghost069, 2019-01-18:
> In #99 someone described a similar issue. I still don't know if colorama was really causing this, but he said he could fix it by upgrading colorama.
> 
> If you pull the latest code, a minimum required colorama version is required and should be installed. Don't know if this helps, but you can try.

Oh my god!It solved.That's the reason! Thank you!

closed 2019-01-18 by ghost069