MichaelGrupp/evo #129: Compare and align the ORB-SLAM2 trajectory with groundtruth

joejevons1001, 2019-01-18:
Hello,

I am running ORB-SLAM2 monocular with my own data. I got the KeyFramTrajectory.txt result and plotted the graph with this evo package.

Now I want to compare and align it with my own GPS data. My GPS data only have (timestamp, longitude and latitude). I got to know there are some functions which can do alignment here. But as my groundtruth data is not in tum format. Is it possible here with my case?

Thank you so much!

MichaelGrupp, 2019-01-21:
If you know the math to do the coordinate conversion, you can convert your latitude/longtitude data to X/Y.

You can then save this 2D position embedded into a 3D pose in TUM format, where the Z-component and rotation quaternion would be the identity (`timestamp x y 0 0 0 0 1`). There is redundant data in this case, but you can just use all the tools that use the TUM format without modification. Probably it makes sense to project also the ORB-SLAM trajectory down to 2D positions.

joejevons1001, 2019-01-22:
Hello, MichaelGrupp,

Thanks for your suggestion. I was also thinking about this. 

But since I am using monocular, the scale is unknown, and the plotted result trajectory is not strictly on Xoy plane, which leans a lot in Z axis, although the trajectory is almost in a same plane. If Z-component is ignored in the GPS data, how could it be aligned?

MichaelGrupp, 2019-01-25:
> Did you mean I convert the GPS to X/Y coordinate and then replace the x y in the trajectory result while keeping all other components unchanged, which can then be saved as my reference data?

I don't understand what you mean by "replacing". My proposal was to convert your timestamp+lat/long GPS data to timestamp+X/Y and save it into the TUM format so you can pass it to `evo_*`.

> But in my GPS data, there are a lot more timestamps and values than the trajectory result. There are 1360 rows in KeyFramTrajectory.txt but over 3000 rows in my GPS data. How to align this?

evo's alignment first matches the trajectory poses based on their timestamps, and then keeps only the matched poses. 

> AND Is it possible to align them in 2D positions since my GPS is 2D but the trajectory is 3D?

You can project both down to 2D if this makes sense to you.

joejevons1001, 2019-01-28:
Thank you!

I misunderstood your first reply.

Now it seems clear to me. I will try with that.

closed 2019-01-31 by MichaelGrupp

joejevons1001, 2019-03-11:
Hi,

I was able to convert my GPS data to the TUM format. But when I run the alignment command,
`evo_ape tum groundtruth.txt KeyFrameTrajectory.txt -as --plot`
 there is an error:
`[ERROR] found no matching timestamps between groundtruth.txt and KeyFrameTrajectory.txt`

The timestamps in my two txt files are not same, the intervals of timestamps are different as well. Are they supposed to be same to use it? How to align the trajectory just based on the geography shape?

Thanks!

MichaelGrupp, 2019-03-11:
> The timestamps in my two txt files are not same, the intervals of timestamps are different as well. Are they supposed to be same to use it?

Timestamps are used to establish point correspondence (registration) for the alignment. They don't need to be the same but in the same interval and ideally synchronized. You can play around with the `--t_offset` an `--t_max_diff` parameters, see `evo_ape tum --help`:

```
  --t_max_diff T_MAX_DIFF
                        maximum timestamp difference for data association
  --t_offset T_OFFSET   constant timestamp offset for data association
```

> How to align the trajectory just based on the geography shape?

This is possible if you use a format without timestamps (e.g. the `kitti` format) and both files have exactly the same number of corresponding (!) poses.

joejevons1001, 2019-03-11:
Hi MichaelGrupp,

Thank you so much for your help. But I am still a bit confused. I played with the --t_offset and --t_max_diff by changing different parameters, still got same error: found no matching.

My estimated trajectory has 1360 poses while my GPS has 3051 poses. The timestamps and intervals are all different at each file and each pose.

Here is my data for your reference. I am appreciated for any help. 
[groudtruth.txt](https://github.com/MichaelGrupp/evo/files/2952801/groudtruth.txt)
[KeyFrameTrajectory.txt](https://github.com/MichaelGrupp/evo/files/2952805/KeyFrameTrajectory.txt)

MichaelGrupp, 2019-03-11:
You are missing time synchronization. The ground truth timestamps start at "0", while your other trajectory uses UNIX timestamps (probably ROS time). See `evo_traj tum groudtruth.txt KeyFrameTrajectory.txt -v
`.

You need to convert them to the same format. E.g. you need to know at which UNIX timestamp your ground truth starts and then add this offset to it. But it's also possible to get something when you subtract the first timestamp from the estimated trajectory:
```
evo_ape tum groudtruth.txt KeyFrameTrajectory.txt -as --t_offset -1547649597
```
But this will produce wrong results unless the real, correct offset is known.

joejevons1001, 2019-03-12:
Hi,

Thanks for the help. I checked the UNIX timestamp conversion issues. The UNIX timestamp is based on the exact date and time.

My groundtruth GPS data was directly acquired from GNSS firmware, how could I get to know the exact time and date when it started since it is counted from 0s? And the KeyFrameTrajectory.txt is directly acquired from ORB-SLAM2 ROS Mono. How could I synchronize the ground truth time with the ROS time?
 
In this case, assuming the ground truth also started from 1547649596 by adding 1547648596 to each timestamps, the offset and result are the same as I directly subtract --t_offset -1547649596 as you mentioned.

Could you give some suggestions based on this situation? Is it still possible to align without any more information.

MichaelGrupp, 2019-03-12:
> My groundtruth GPS data was directly acquired from GNSS firmware, how could I get to know the exact time and date when it started since it is counted from 0s?

I can't help you with that. This depends on how you recorded your stuff.

> And the KeyFrameTrajectory.txt is directly acquired from ORB-SLAM2 ROS Mono. How could I synchronize the ground truth time with the ROS time?

Same here. Ground truth and estimated trajectory need to be in the same time dimension.

> Is it still possible to align without any more information.

Not based on timestamp registration (unless you guess the offset). Maybe you can use an ICP-like algorithm that establishes point correspondences in some other way.

barloccia, 2019-10-25:
Hi, 
Thank you for your great work. Reading at the question i noticed that you suggest to project 3D trajectory onto a 2D plane. I am in the same situation where i should compare GPS data with SLAM result and, of course, GPS data are in 2D. As you suggested i saved the GPS in the same format as tum (timestamp x y 0 0 0 0 1) and it makes me sense. The comparison is working well, the only difference is that GPS data is a 'flat' trajectory, while SLAM data are different values on Z (and it also is ok). Finally my question: since the SLAM data have a random reference system, how is possible to project the 3D data onto the same 2D plane of GPS? 
I think when you align you basically find out rotation and translation matrices to retrieve the same reference system of GPS (in general of the reference trajectory). Is that correct?

MichaelGrupp, 2019-10-28:
> I think when you align you basically find out rotation and translation matrices to retrieve the same reference system of GPS (in general of the reference trajectory). Is that correct?

The type of alignment that we use in this library is trying to find a transformation which makes the two set of points fit together as good as possible. This is fine for comparisons but it's not bringing the data into the same reference system nor doing any geo-referencing.

barloccia, 2019-10-29:
Ok, thank you for the reply.
The thing is that i have a RGB-D system so that my comparison between ground-truth (GPS) and estimated should be done without any modification of the trajectories. I just want to align the first point and see what happens, starting from a common origin (0,0,0) for both the trajectories. I have the same numbers of points and correspondences.
If i try 'align_origin' i obtain something as in the pictures. It seems that the reference is not the same, or something like that, i am a bit confused. Do you have any idea about that?
[image]
[image]

barloccia, 2019-10-30:
Yes is what i am checking right now. I can say that they don't have the same axis convention, so  maybe this is the reason. I will try to change the order and the sign of the axis and see if it woks.

khyaph, 2020-02-26:
Hey, did you find the reason? I have data from colmap and orbslam and the same problem. With --n_to_align X I get a useful result, however I can't use this command with evo_ape. 

And a more general question:
Does align_origin only use the first data point of each trajectory? Is it the same as the transformation_left when I use the differences between the first point of each trajectory?

MichaelGrupp, 2020-02-26:
> Does align_origin only use the first data point of each trajectory? Is it the same as the transformation_left when I use the differences between the first point of each trajectory?

Almost, it does also timestamp matching first because the trajectories might not start at exactly the same time.