MichaelGrupp/evo #133: Need help with running the code

adarshsehgal, 2019-01-30:
Hi,

I installed the code as mentioned in the readme and am now trying to run the commands on test data provided. I tried:

evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
evo_ape kitti KITTI_00_gt.txt KITTI_00_ORB.txt -va --plot --plot_mode xz --save_results results/ORB.zip

But looks like none of the commands are doing anything for me. Can you please tell me if I am doing anything wrong?

Thank you

MichaelGrupp, 2019-01-30:
Please provide more information if you can. Ideally, fill out the [issue template](https://github.com/MichaelGrupp/evo/blob/master/.github/ISSUE_TEMPLATE/error-report.md).

MichaelGrupp, 2019-01-30:
Currently I can only say it might be related to this issue: https://github.com/MichaelGrupp/evo/issues/128

adarshsehgal, 2019-01-30:
**Description:**
It shows nothing after I run the test command. I also don't see any file in the results folder when I run the command. I had looked at issue #128 and ugraded colorama to version 0.4, but that didn't help.

**Command:**
```
evo_ape kitti KITTI_00_gt.txt KITTI_00_ORB.txt -va --plot --plot_mode xz --save_results results/ORB.zip
```

**Console output:**
[image]

I do not see any output on the console, nor the files in the results folder. I did create results folder before running this command. 

Please give also the following information:
* evo version number shown by `evo pkg --version`: 
* Python version shown by `evo pkg --pyversion`: 
* operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* did you change the source code? (yes / no): 
* output of `evo_config show --brief --no_color`: 
```
v1.3.2
3.5.2
Ubuntu 16.04
no
{
    "console_logging_format": "%(message)s",
    "global_logfile_enabled": false,
    "plot_backend": "Qt5Agg",
    "plot_export_format": "pdf",
    "plot_figsize": [
        6,
        6
    ],
    "plot_fontfamily": "sans-serif",
    "plot_fontscale": 1.0,
    "plot_invert_xaxis": false,
    "plot_invert_yaxis": false,
    "plot_linewidth": 1.5,
    "plot_multi_cmap": "none",
    "plot_reference_alpha": 0.5,
    "plot_reference_color": "black",
    "plot_reference_linestyle": "--",
    "plot_seaborn_palette": "deep6",
    "plot_seaborn_style": "darkgrid",
    "plot_split": false,
    "plot_texsystem": "pdflatex",
    "plot_trajectory_alpha": 0.75,
    "plot_trajectory_cmap": "jet",
    "plot_usetex": false,
    "plot_xyz_realistic": true,
    "save_traj_in_zip": false,
    "table_export_data": "stats",
    "table_export_format": "csv",
    "table_export_transpose": true
}

```

MichaelGrupp, 2019-01-31:
Thanks. I created a virtual environment [like here](https://github.com/MichaelGrupp/evo/blob/master/doc/install_in_virtualenv.md) for Python 3.5 and installed evo with `pip3 install evo --no-binary evo --upgrade`.

However, everything works fine for me, so I can't reproduce what you see here...

MichaelGrupp, 2019-02-01:
Weird. But good to hear that!