MichaelGrupp/evo #137: Multiple run averaging

Kallista, 2019-02-07:
If one were to go about averaging multiple runs of the same algorithm to evaluate against GT and eventually against other SLAM/odometries, what would be a proper way to do it? 

Some SLAM/odometry algorithms, especially pure monocular, are non-deterministic and it would sometimes make sense to average ~10 runs and only then do comparisons.

MichaelGrupp, 2019-02-07:
You can save the result of each run and then average.

If you don't want to code some custom Python script for that, you could also just run a simple bash script:
```bash
#!/bin/bash
set -e

mkdir -p myresults

for i in {1..10}; do
    evo_rpe kitti KITTI_00_gt.txt KITTI_00_ORB.txt --save_results myresults/${i}.zip
done

evo_res myresults/* --use_filenames --save_table table.csv
```

Then you can open the table file with your favorite spreadsheet program and average the values.

Kallista, 2019-02-07:
Ok, thanks for the example!

But the resulting csv data is only enough to draw f.e. the stats page of evo_res, but not enough for a boxplot/histogram? I'll have to dump the data then on my own to get those plots? (and a trajectory plot isn't too useful for the effort)

MichaelGrupp, 2019-02-08:
This Python script averages the stats, combines the raw error values and writes a new file:

```python
#!/usr/bin/env python

import glob
import os

import numpy as np

from evo.core import result
from evo.tools import file_interface

merged_result = result.Result()

i = 0
for res_file in glob.glob("./*.zip"):
    result = file_interface.load_res_file(res_file)
    if i == 0:
        merged_result.info = result.info
        merged_result.stats = result.stats
        merged_result.np_arrays = result.np_arrays
    else:
        merged_result.stats = {
            key: ((merged_result.stats[key] + result.stats[key]) / 2)
            for key in merged_result.stats
        }
        # only works if raw value arrays have exact same length
        merged_result.np_arrays = {
            key: np.mean(
                np.array([merged_result.np_arrays[key],
                          result.np_arrays[key]]), axis=0)
            for key in result.np_arrays
        }
        # alternative: just append, lengths can be different
        # merged_result.np_arrays = {
        #     key: np.append(merged_result.np_arrays[key], result.np_arrays[key])
        #     for key in result.np_arrays
        # }
    i += 1

if not os.path.exists("merged"):
    os.makedirs("merged")
file_interface.save_res_file("./merged/merged_result.zip", merged_result)
```

MichaelGrupp, 2019-02-08:
The tricky part is averaging the raw error values arrays, which should have the exact same length.

But if you only care about box plot / stats, they can also be just concatenated like in the part I commented out. Then the file gets also bigger of course.

Kallista, 2019-02-11:
Wow thanks, I'll try that out!

MichaelGrupp, 2019-02-20:
Keeping this open as it might be a feature that could be added to `evo_res`.

commit 36a0133c (2019-03-16): Add result merging feature. (#150)

Closes #137.