MichaelGrupp/evo #144: Kitti metric

andre-nguyen, 2019-02-14:
Hi there,

I'm trying to use evo_rpe to compute the same translational error metric the kitti odometry benchmark as presented in their website results. My understanding is to get the same thing we would need to do

```
evo_rpe [format] [ref] [est] -va -d [length] -u m --all_pairs
```
where `[length] = {100, 200, 300, 400, 500, 600, 700, 800}`. We should then normalize all the mean errors w.r.t. the length and the average out these results. Can someone confirm?

Thanks for the great software!

BrunoEduardoCSantos, 2019-02-15:
`evo_rpe kitti data/kitti/dataset/poses/00.txt  python_scripts/data/odometry/lidar_kitti.txt --pose_relation full --delta 100 --delta_unit m 
RPE w.r.t. full transformation 
for delta = 100.0 (m) using consecutive pairs
(not aligned)

       max	199.251959
      mean	109.047650
    median	146.064418
       min	9.655466
      rmse	129.760700
       sse	606162.214747
       std	70.330998
`
Is it normal to obtain such large values?
[image]

Thanks,
Bruno

andre-nguyen, 2019-02-15:
What if you use only the translation? (Remove pose relation parameter)

BrunoEduardoCSantos, 2019-02-15:
Hi @andre-nguyen ,
I tried only with translation like this:
`evo_rpe kitti data/kitti/dataset/poses/00.txt python_scripts/data/odometry/lidar_kitti.txt   --delta 100 --delta_unit m`
Still the same results.

MichaelGrupp, 2019-02-15:
@BrunoEduardoCSantos Would be easier to analyze if you could upload your two files, so we could have a look.

BrunoEduardoCSantos, 2019-02-15:
Sure. Thanks a lot for your help. 
[lidar_kitti_2.txt](https://github.com/MichaelGrupp/evo/files/2869797/lidar_kitti_2.txt)
[00.txt](https://github.com/MichaelGrupp/evo/files/2869799/00.txt)

MichaelGrupp, 2019-02-15:
The `lidar_kitti_2.txt` trajectory has jumps. It looks like the trajectory contains the jumps that occur when loop closures happen, as seen from a global fixed frame. Do you save this trajectory after all optimizations are done or do you record it "live" (and record potential loop closure jumps)?

[image]

BrunoEduardoCSantos, 2019-02-15:
Thanks @MichaelGrupp  for your feedback. Currently, I am recording "live" . Currently, I am recording odometry from SLAM algorithm in ROS.

andre-nguyen, 2019-02-15:
@MichaelGrupp Can you comment on my original question?

MichaelGrupp, 2019-02-16:
Your approach looks reasonable. However, if you want to get exactly the same results I would recommend to have a look at the KITTI benchmark's official code. You can download this from the page where the results are published ("development kit"). It's one C++ file if I remember correctly.

closed 2019-02-17 by andre-nguyen

BrunoEduardoCSantos, 2019-02-18:
Hi @MichaelGrupp ,
Do you think could be an issue with loop closure?
Thanks,
Bruno

MichaelGrupp, 2019-02-20:
I would say it's better to save the fully optimized trajectory after all computations are done instead of recording it for your use case, if possible.

BrunoEduardoCSantos, 2019-02-21:
Thanks @MichaelGrupp  for your feedback.

mnik17, 2019-08-01:
@MichaelGrupp Hi, I'm going to post in this thread, since my question is related. 

Going by your answers here, is it correct to assume, that the values, which are output in the console are not normalized wrt. to the delta? For example if I use the command:

`evo_rpe [bag] [mybag.bag] [ref] [est] --delta 100 --delta_unit m --pose_relation trans_part`

and let's say the mean is 10. Is that value normalized wrt. the delta i.e. 100m or should I do this manually? 

Thanks in advance!

MichaelGrupp, 2019-08-02:
It's not normalized with respect to the delta value.

lgl-cqu, 2025-04-07:
> [@MichaelGrupp](https://github.com/MichaelGrupp) Hi, I'm going to post in this thread, since my question is related.
>
> Going by your answers here, is it correct to assume, that the values, which are [...]

Hi there. I am also curious about this question, would you please tell me the correct solution if you confirmed the correct operation?