MichaelGrupp/evo #152: Scale of Mono-KITTI

liangsongLee, 2019-03-17:
Hello, I used orbslam2 to run KITTI in mono mode. And I got the trajectory .txt file of keyframes in the form of TUM. Then I use --save_as_kitti to transform the .txt file to KITTI's form. Finally, I use   evo_traj kitti KeyFrameTrajectory.txt --ref=ground_truth.txt -p --plot_mode=xz .  I got a plot without align. However, I dont know how to align it. Please help me , thank you very much.
[image]

MichaelGrupp, 2019-03-17:
See here: https://github.com/MichaelGrupp/evo/wiki/evo_traj#geometric-alignment

So you could try this:
```
 evo_traj kitti KeyFrameTrajectory.txt --ref=ground_truth.txt -p --plot_mode=xz --correct_scale --align
```

liangsongLee, 2019-03-17:
Thank you, I tried this, but I got the error : data matrices must have the same shape

MichaelGrupp, 2019-03-17:
Can you upload your data here?

liangsongLee, 2019-03-17:
OK, here are the datas.
[ground_truth.txt](https://github.com/MichaelGrupp/evo/files/2975089/ground_truth.txt)
[KeyFrameTrajectory.txt](https://github.com/MichaelGrupp/evo/files/2975090/KeyFrameTrajectory.txt)

liangsongLee, 2019-03-17:
And this is the file that I got from mono_kitti of ORBSLAM2 , which is in the form of TUM data before I saved it as kitti use your code.
[KeyFrameTrajectory_mono.txt](https://github.com/MichaelGrupp/evo/files/2975100/KeyFrameTrajectory_mono.txt)

liangsongLee, 2019-03-17:
And this is the trajectory file of sequence 09 of KITTI that I got from mono_kitti of ORBSLAM2 , which is in the form of TUM data before I saved it as  form of kitti using your code here.
[KeyFrameTrajectory_mono.txt](https://github.com/MichaelGrupp/evo/files/2975103/KeyFrameTrajectory_mono.txt)

MichaelGrupp, 2019-03-17:
The problem is that the `kitti` format doesn't have timestamps to do timestamp correspondence for alignment, so the only way to do it is that the number of poses must be exactly the same with this format. Formats with timestamps like `tum` don't have this problem.

But you can probably solve that. The KITTI dataset also has separate files that provide the timestamps. 
From a similar issue https://github.com/MichaelGrupp/evo/issues/125 we wrote a script that can combine this timestamp file and a `kitti` poses file to a proper trajectory. You can download it here and use it to convert the ground truth to `tum` format:
https://github.com/MichaelGrupp/evo/blob/master/contrib/kitti_poses_and_timestamps_to_trajectory.py

liangsongLee, 2019-03-17:
OK, got it. thank you Michael.

closed 2019-03-18 by liangsongLee

ashwinjoseph95, 2024-08-01:
Hi guys had a query on the same topic,
I was using VINS-FUSION Stereo Visual SLAM to get odometry predictions. Wanted to use it with evo for evaluating the performance using Ground Truth for KITTI odometry sequence 00 file.
I suppose the timestamps are in nanoseconds and I was able to geenrate the timestamped version of kitti groudn truth file in tum version using the script [kitti_poses_and_timestamps_to_trajectory.py](https://github.com/MichaelGrupp/evo/blob/master/contrib/kitti_poses_and_timestamps_to_trajectory.py) which Michael has provided. I have got the predicted trajectory in a file named vio.csv from VINS-FUSION for the same KITTI dataset. The number of sequences do not match like there is a difference of about 18 rows. I want to understand that is it just ok to delete few rows anywhere to match the count on both set of files? And in that case when absolute error metric won't it get affected as clearly its not apparantly the same timestamp odometry position difference.