MichaelGrupp/evo #153: Convert from TUM to Kitti

Usama323, 2019-03-24:
I need to ask a question, which function is used to convert from TUM to KITTI data set ? 

Can you kindly tell me which line is used to convert it? 
Also nominate portion of code which is used to convert from TUM format to KITTI format (4x4 matrix)

MichaelGrupp, 2019-03-24:
There is no direct function. But with `evo.tools.file_interface` you can load the trajectory with `read_tum_trajectory_file` and then pass it to `write_kitti_poses_file`.

You can follow the code there, in the end the conversion from quaternion to rotation matrix is done in the transformations module.

Usama323, 2019-03-25:
Thanks @MichaelGrupp for guideline, I would check this out !!

Usama323, 2019-03-25:
Thank you for fast answer, I need to convert Quaternion to rotation matrix (4x4 matrix) . Kindly tell the function which is used to do this task ? Nominate line number or python file for this task.  
I am asking this because there is confusion in this conversion across multiple platform, like in following example I have this Quaternion Matrix 
Q = [0.6604 0.6064 -0.2939 -0.3313];   (Ox =0.6604, Oy=0.6064, Oz=-0.2939 & Ow=-0.3313)

gives me following Rotation matrix when I used built-in function of MATLAB (quat2dcm)
R = [0.6077 -0.799 -0.0136; 0.0811 0.045 0.09957; -0.7900 -0.6062 0.0918] 

gives me following Rotation matrix when I used ROS function (tf)
R = [0.0917 0.606 -0.7899; 0.9956 -0.045 0.811; 0.0136 -0.794 -0.6077]

MichaelGrupp, 2019-03-25:
You can find the code I use in `evo/core/transformations.py` (`quaternion_matrix()`). It gives 4x4 rigid body transformation matrix that includes a 3x3 rotation matrix. There is no "4x4 rotation matrix" in 3d.

The Matlab thingy you wrote doesn't look like correct, it converts to a direction cosine matrix.

MichaelGrupp, 2019-03-25:
I recommend looking up the math behind that conversion to verify the function.

closed 2019-03-25 by MichaelGrupp

Usama323, 2019-03-25:
Thanks a lot for helping me out @MichaelGrupp