MichaelGrupp/evo #154: about mono camera pose result evaluate

Chrislzy1993, 2019-03-25:
hi, @plusk01  @MichaelGrupp  @ToniRV  @slowlythinking  @miseryofdevice i have a problem!
when i run orbslam in mono mode by kitti dataset,  i get a keyframe result, i want to evaluate the result with groundtruth pose.  cause mono mode lack of scale,  when i use evo to draw the trajectory, i get the following pic: 
and keyframe num is about 2000 , and camera frame is about 4000
[image]
[image]
[image]
i am confused, how to deal with scale problem, i am a Newbies

MichaelGrupp, 2019-03-25:
Please check if this helps: https://github.com/MichaelGrupp/evo/issues/152

Chrislzy1993, 2019-03-25:
@MichaelGrupp , thanks, i will try it!

Chrislzy1993, 2019-03-26:
thanks, it works!

closed 2019-03-26 by Chrislzy1993

CLM666, 2019-09-08:
@Chrislzy1993 
hi, Chrisizy. I have the same problem about kitti format, can you tell me how to use the kitti_poses_and_timestamps_to_trajectory.py to convert data to tum format?
I have used this command:
python2 /home/clm/evo/contrib/kitti_poses_and_timestamps_to_trajectory.py poses_file=./00.txt timestamp_file=./times.txt trajectory_out=./result
reminding me : evo.tools.file_interface.FileInterfaceException: csv file poses_file=./00.txt does not exist
 I am confused, please help me if you have time!  Thank you very much!