MichaelGrupp/evo #155: How to record KITTI pose files?

ywyue, 2019-04-02:
Hello, I am testing LOAM using KITTI odometry data, I used the command "$rostopic echo /laser_odom_to_init/pose/pose > sequence_06.txt"  to get the result file. Then I used "evo_traj kitti sequence_06.txt --ref=KITTI_06_gt.txt -p --plot_mode=xz", the error is "KITTI pose files must have 12 entries per row and no trailing delimiter at the end of the rows (space)". 
The sequence_06.txt is organized as below. 
[image]
Could anyone please tell me how to modify the command to get the right result file?

MichaelGrupp, 2019-04-02:
This obviously doesn't work: `rostopic echo` doesn't print in KITTI format (in your case it looks like it's printing `geometry_msgs/Pose` messages.

I would recommend to write a small ROS subscriber node that listens to the topic and converts & writes messages to the final file format. The KITTI format is described here: https://github.com/MichaelGrupp/evo/wiki/Formats#kitti---kitti-dataset-pose-format

ywyue, 2019-04-02:
Thanks for your reply. I wonder if there are some tools that can convert geometry_msgs/Pose to KITTI format data?

MichaelGrupp, 2019-04-02:
Probably not. evo supports a few ROS formats, but not `geometry_msgs/Pose` at the moment. But you only need to convert the xyz-position and the quaternion of `geometry_msgs/Pose` to a 4x4 transformation matrix and write the first 3 rows to the file.

If you're lazy you can keep the translation and quaternion as they are and write them to a TUM file with dummy timestamps and then finally use `evo_traj --save_as_kitti ...` to convert it.

ywyue, 2019-04-02:
Thanks a lot for your help! I will try it.

ywyue, 2019-04-04:
Hello. I kept the translation and quaternion as they are and wrote them to a TUM file with same timestamps (for example the all timestamps were set as '1'). Then I used 'evo_traj tum 06.csv --save_as_kitti' to convert it to KITTI format. Finally, I used 'evo_traj kitti 06.kitti --ref=kitti_06_gt.txt -p --plot_mode=xz' to compare the trajectory. However, I got the symmetrical trajectory. Does anybody know why? Thanks a lot！
[image]

MichaelGrupp, 2019-04-04:
Maybe your SLAM is using a different coordinate convention where the Z axis is flipped?

You can also just use alignment in evo.

MichaelGrupp, 2019-04-10:
Closing because it's answered

closed 2019-04-10 by MichaelGrupp

xuwuzhou, 2019-12-01:
> Hello. I kept the translation and quaternion as they are and wrote them to a TUM file with same timestamps (for example the all timestamps were set as '1'). Then I used 'evo_traj tum 06.csv [...]

hello,where the the translation and quaternion you subscribed in source code,I can't find it,I guess that they may be in the laserOdometry.cpp,but indeed they are not there.