MichaelGrupp/evo #156: ValueError: deep6 is not a valid palette name

Adamquhao, 2019-04-10:
thanks for making your work public. recently i run the first example 'Plot multiple trajectories'. however, it didn't work. the error details are shown as following:

qu@qu-TM1613:~/work/evo/test/data$ evo_traj kitti KITTI_00_ORB.txt KITTI_00_SPTAM.txt --ref=KITTI_00_gt.txt -p --plot_mode=xz
--------------------------------------------------------------------------------
name:	KITTI_00_ORB
infos:	4541 poses, 3705.098m path length
--------------------------------------------------------------------------------
name:	KITTI_00_SPTAM
infos:	4541 poses, 3718.501m path length
--------------------------------------------------------------------------------
name:	KITTI_00_gt
infos:	4541 poses, 3724.187m path length
[ERROR] Unhandled error in evo.main_traj
Traceback (most recent call last):
  File "/home/qu/anaconda2/lib/python2.7/site-packages/evo/entry_points.py", line 90, in launch
    main_module.run(args)
  File "/home/qu/anaconda2/lib/python2.7/site-packages/evo/main_traj.py", line 342, in run
    from evo.tools.plot import PlotMode
  File "/home/qu/anaconda2/lib/python2.7/site-packages/evo/tools/plot.py", line 50, in <module>
    font_scale=SETTINGS.plot_fontscale)
  File "/home/qu/anaconda2/lib/python2.7/site-packages/seaborn/rcmod.py", line 110, in set
    set_palette(palette, color_codes=color_codes)
  File "/home/qu/anaconda2/lib/python2.7/site-packages/seaborn/rcmod.py", line 504, in set_palette
    colors = palettes.color_palette(palette, n_colors, desat)
  File "/home/qu/anaconda2/lib/python2.7/site-packages/seaborn/palettes.py", line 180, in color_palette
    raise ValueError("%s is not a valid palette name" % palette)
ValueError: deep6 is not a valid palette name

[ERROR] evo module evo.main_traj crashed - no logfile written (disabled)

MichaelGrupp, 2019-04-10:
You probably don't use the latest version of seaborn, so you could upgrade this using your Python package manager.

Alternatively, `evo_config set plot_seaborn_palette deep` should also work.

Adamquhao, 2019-04-10:
thanks q very much, i have addressed this problem by your suggestion!!!

closed 2019-04-10 by MichaelGrupp